klogg/fl2000_drm #6: Compilation failed on linux kernel 5.1.6

tinytangent, 2019-06-05:
Error message for command `make` is:

> /home/tinytangent/fl2000_drm/fl2000_drm.c: In function ‘fl2000_bind’:
/home/tinytangent/fl2000_drm/fl2000_drm.c:222:8: error: implicit declaration of function ‘drm_fb_cma_fbdev_init’; did you mean ‘drm_fb_helper_init’? [-Werror=implicit-function-declaration]
  ret = drm_fb_cma_fbdev_init(drm, BPP, MAX_CONN);
        ^~~~~~~~~~~~~~~~~~~~~
        drm_fb_helper_init
/home/tinytangent/fl2000_drm/fl2000_drm.c:228:2: error: implicit declaration of function ‘drm_kms_helper_poll_init’; did you mean ‘drm_fb_helper_defio_init’? [-Werror=implicit-function-declaration]
  drm_kms_helper_poll_init(drm);
  ^~~~~~~~~~~~~~~~~~~~~~~~
  drm_fb_helper_defio_init
/home/tinytangent/fl2000_drm/fl2000_drm.c: In function ‘fl2000_unbind’:
/home/tinytangent/fl2000_drm/fl2000_drm.c:254:2: error: implicit declaration of function ‘drm_fb_cma_fbdev_fini’; did you mean ‘drm_fb_helper_fini’? [-Werror=implicit-function-declaration]
  drm_fb_cma_fbdev_fini(drm);
  ^~~~~~~~~~~~~~~~~~~~~
  drm_fb_helper_fini

Maybe the kernel API is changed?

klogg, 2019-09-06:
Most probably, patches are welcome. I am working on Ubuntu with 5.0.0 now. Will try to move to next Linux version as soon as finish all functions implementation

klogg, 2019-09-07:
Just checked the APIs - they are the same. Probably it was something with includes in that version, current version builds OK

closed 2019-09-07 by klogg