klogg/fl2000_drm #13: Which Linux kernels are compatible with fl2k_drm?

borvelt, 2020-01-06:
I have `Linux 5.0.0-37-generic` but I will face with error after `make` command.

```~/fl2000_drm/fl2000.h:36:10: fatal error: drm/drm_probe_helper.h: No such file or directory```

My Linux kernel doesn't contain this `drm_probe_helper.h`, should I update my kernel?

borvelt, 2020-01-07:
I've changed KSRC to new kernel address and now I have this issues:
```
./arch/x86/include/asm/bug.h:35:22: error: expected identifier or ‘(’ before string constant
  asm_inline volatile("1:\t" ins "\n"
```
and some thing like this: 
```
./include/drm/drm_atomic_helper.h:221:2: note: in expansion of macro ‘WARN_ON’
  WARN_ON((new_plane_state->crtc == NULL && new_plane_state->fb != NULL) ||
  ^~~~~~~
```

any body have idea?

borvelt, 2020-01-07:
@klogg

klogg, 2020-05-05:
Sorry for not reacting promptly @borvelt 
Currently I build it on 5.4 with Ubuntu 20.04

closed 2020-10-14 by klogg