klogg/fl2000_drm #14: Compilation failed on Linux 5.7

TuxThePenguin0, 2020-05-04:
Make fails with:
`fl2000_drm/fl2000.h:28:10: fatal error: drm/drmP.h: No such file or directory`

It seems as though drmP was replaced in a recent kernel version, seems as early as 5.5

klogg, 2020-05-05:
Yeah, I build it for 5.4 now with Ubuntu 20.04. As soon as I resolve
https://github.com/klogg/fl2000_drm/issues/15
https://github.com/klogg/fl2000_drm/issues/16
I will move to cleanup and possible upstream support. Before that - patches are welcome

fcrohas, 2020-05-06:
Hi,
  I just read your comment today, i'm on opensuse thumbleweed with kernel 5.6.4, mostly the same issue. I forked you repo and started to push my change to get it compile. But unfortunally, it compile ,load but can't check as my dongle is vga output. Bought it a while ago to try osmofl2k driver :)

fcrohas, 2020-05-08:
Ok i ve done the changes to support latest kernel. Now i look how to add dumb_vga_dac via devicetree ....

TuxThePenguin0, 2020-05-11:
I have a HDMI version of the adapter and your changes compile and load fine on my machine, however I'm not sure how to test it. If someone knows how then help would be appreciated

klogg, 2020-05-12:
I test with virtme/qemu, check out the DEBUG.md and scripts folder. I have solved the issue with PLL and buffer byteswap, will move to continue cleanup later this week.
Patches are welcome 😁

klogg, 2020-11-01:
Migrating to 5.8 kernel with Ubuntu 20.10

closed 2020-11-01 by klogg