klogg/fl2000_drm #23: No Display, Sound and Network stops working

DumiM, 2020-09-05:
Hi,

Firstly, I want to say that I really appreciate this project. **Cheers to the developer!**
Secondly, I want to point out that there really isn't clear instructions on how to install and use the driver (the section _How to use_ in the README is not helpful) for someone who isn't knowledge about drivers, like myself. If that information could be extended that will be appreciated.
Thirdly, I tried installing the drivers but the display is not picked up in the display settings and the screen is blank. However, now I have two new options in the Sound settings - `usb extscreen` _Digital_ and _Analogue_. Also, my network breaks and the internet is not accessible (I presume it thinks of it as an ethernet connection perhaps?)

I am using Ubuntu 20.04. To install I used these steps:
1. `make all`
2. `sudo ./startup.sh`  in the scripts folder. This did not work - after some digging, I changed the tcpdump line to `tcpdump -Z root -i usbmon2 -nn -w /tmp/usb.pcap -c 20000 &` and moved the file to the main folder. Following this, it worked and have the response : `listening on usbmon2, link-type USB_LINUX_MMAPPED (USB with padded Linux header), capture size 262144 bytes
`. But the problems I stated above still remains.

Any help would be really appreciated. Thanks

DumiM, 2020-09-25:
@klogg Hi! Is this repo still being monitored? Would mean a lot if you could have a look at the issue mentioned above.
Also, is there a donation page for this project, because I would love to show my support for this!

klogg, 2020-09-25:
Hi, I am working on update, will get back to you soon
Sorry for not following up promptly

klogg, 2020-09-26:
@DumiM  JFYI - right now I am performing most of testing in a virtualized environment - passing whole USB controller as a PCI device into a qemu VM (using vfio) and while there are known "remove" issues (#12 and #22) I am concentrating on "happy paths" for small (480x800) and big (1920x1080) displays due to differences in memory allocation for them. Right now I have 2 issue left, both are on larger device: a) console does not start (something with fbdev emulation) and b) when playing kmscube some frames are getting lost. I hope to fix them soon and then I'll switch to testing in non-virt environment (Ubuntu 20.04 with XFCE).

It would be helpful if you can describe your system:
 - Your target HW (CPU / RAM)
 - Distro you're using & version
 - kernel version
 - USB controller you have in your system
 - your display resolution and clock

DumiM, 2020-09-27:
@klogg Thanks mate that's awesome.
Here is my system info
- My CPU: Intel i7-6500U CPU @ 2.50GHz × 4 
- RAM: 8 GB
- Distro: Ubuntu 20.04
- Kernel: 5.4.0-47-generic
- Laptop Display: 1920x1080 
- External Monitor Display: 1440x900 @ 106.5 pixel clock
Hope this helps

klogg, 2020-09-27:
Thanks
Please rebuild the driver with latest changes, and try to reproduce the issue. After trying to run the driver please give me output of `dmesg` command in console

klogg, 2020-09-28:
@DumiM I suggest to switch to Gitter for discussion, this thread is getting rather active

DumiM, 2020-09-28:
@klogg Thanks will do. Will continue this on Glitter.

klogg, 2020-09-29:
Closing this issue as it turned out to be reported from a system with on non-FL2000 dongle

closed 2020-09-29 by klogg