klogg/fl2000_drm #33: Problems with the IT66121 driver

Riiquezz, 2020-10-29:
The problems that are occurring are as follows.

1. When you insert the driver, it completely locks the system and it is necessary to shut down the computer by means of brute force.
I think the third time this happened to me, my system became corrupted and it was necessary to do a restoration.

2. If the module is inserted and does not lock, when I put the computer in sleep mode and come back, the system is all locked too, having to shut down via brute force.

klogg, 2020-10-29:
Managed to reproduce on a system without virtme/qemu
Most probably issue with IT66121 interrupt processing

klogg, 2020-10-31:
Seems like it is not related to IT66121 interrupt but rather to issue of pages allocation in streaming module:
```
ct 31 22:53:47 Blacktower kernel: [  110.785425] usb 2-4: Mode requested:  "": 0 27000 720 732 796 864 576 581 586 625 0x0 0xa
Oct 31 22:53:47 Blacktower kernel: [  110.785426] usb 2-4: Mode configured: "": 0 27000 720 732 796 864 576 581 586 625 0x0 0xa
Oct 31 22:53:47 Blacktower kernel: [  110.802973] usb 2-4: Setting AVI infoframe for mode: "": 0 27000 720 732 796 864 576 581 586 625 0x0 0xa
Oct 31 22:53:48 Blacktower kernel: [  111.400785] BUG: unable to handle page fault for address: ffffafaec7130004
Oct 31 22:53:48 Blacktower kernel: [  111.400793] #PF: supervisor write access in kernel mode
Oct 31 22:53:48 Blacktower kernel: [  111.400796] #PF: error_code(0x0002) - not-present page
Oct 31 22:53:48 Blacktower kernel: [  111.400799] PGD 89b55e067 P4D 89b55e067 PUD 89b55f067 PMD 7a38a3067 PTE 0
Oct 31 22:53:48 Blacktower kernel: [  111.400809] Oops: 0002 [#1] SMP PTI
Oct 31 22:53:48 Blacktower kernel: [  111.400815] CPU: 4 PID: 935 Comm: Xorg Tainted: P           OE     5.4.0-52-generic #57-Ubuntu
Oct 31 22:53:48 Blacktower kernel: [  111.400818] Hardware name: System manufacturer System Product Name/MAXIMUS VIII HERO, BIOS 3802 03/15/2018
Oct 31 22:53:48 Blacktower kernel: [  111.400829] RIP: 0010:fl2000_stream_compress+0xda/0x190 [fl2000]
Oct 31 22:53:48 Blacktower kernel: [  111.400834] Code: e4 74 3f 44 8b 06 89 ca 48 83 c6 04 83 f2 04 44 88 04 17 8b 46 fc 44 8d 41 01 8d 51 02 41 83 f0 04 83 f2 04 83 c1 03 0f b6 c4 <42> 88 04 07 44 8b 46 fc 41 c1 e8 10 44 88 04 17 44 39 f1 75 c4 4c
Oct 31 22:53:48 Blacktower kernel: [  111.400838] RSP: 0018:ffffafaec1403a48 EFLAGS: 00010012
Oct 31 22:53:48 Blacktower kernel: [  111.400843] RAX: 0000000000000000 RBX: ffffafaec2c6a000 RCX: 0000000000000402
Oct 31 22:53:48 Blacktower kernel: [  111.400846] RDX: 0000000000000405 RSI: ffffafaec2c6a558 RDI: ffffafaec712fc00
Oct 31 22:53:48 Blacktower kernel: [  111.400849] RBP: ffffafaec1403a88 R08: 0000000000000404 R09: 0000000000000195
Oct 31 22:53:48 Blacktower kernel: [  111.400852] R10: 0000000000000c00 R11: 00000000000003ff R12: 0000000000000400
Oct 31 22:53:48 Blacktower kernel: [  111.400854] R13: 0000000000000300 R14: 0000000000000c00 R15: ffff8f1807520018
Oct 31 22:53:48 Blacktower kernel: [  111.400859] FS:  00007f5923cfba80(0000) GS:ffff8f18deb00000(0000) knlGS:0000000000000000
Oct 31 22:53:48 Blacktower kernel: [  111.400862] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Oct 31 22:53:48 Blacktower kernel: [  111.400866] CR2: ffffafaec7130004 CR3: 0000000874070005 CR4: 00000000003606e0
Oct 31 22:53:48 Blacktower kernel: [  111.400869] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
Oct 31 22:53:48 Blacktower kernel: [  111.400872] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
Oct 31 22:53:48 Blacktower kernel: [  111.400875] Call Trace:
Oct 31 22:53:48 Blacktower kernel: [  111.400887]  fl2000_display_update+0x1a6/0x1b0 [fl2000]
Oct 31 22:53:48 Blacktower kernel: [  111.400912]  drm_simple_kms_plane_atomic_update+0x2a/0x30 [drm_kms_helper]
Oct 31 22:53:48 Blacktower kernel: [  111.400932]  drm_atomic_helper_commit_planes+0xbd/0x220 [drm_kms_helper]
Oct 31 22:53:48 Blacktower kernel: [  111.400951]  drm_atomic_helper_commit_tail+0x2c/0x70 [drm_kms_helper]
Oct 31 22:53:48 Blacktower kernel: [  111.400969]  commit_tail+0xd7/0x110 [drm_kms_helper]
Oct 31 22:53:48 Blacktower kernel: [  111.400985]  drm_atomic_helper_commit+0x118/0x120 [drm_kms_helper]
Oct 31 22:53:48 Blacktower kernel: [  111.401031]  drm_atomic_commit+0x4a/0x50 [drm]
Oct 31 22:53:48 Blacktower kernel: [  111.401050]  drm_atomic_helper_set_config+0x89/0xa0 [drm_kms_helper]
Oct 31 22:53:48 Blacktower kernel: [  111.401083]  drm_mode_setcrtc+0x1cd/0x7a0 [drm]
Oct 31 22:53:48 Blacktower kernel: [  111.401115]  ? drm_mode_getcrtc+0x190/0x190 [drm]
Oct 31 22:53:48 Blacktower kernel: [  111.401143]  drm_ioctl_kernel+0xae/0xf0 [drm]
Oct 31 22:53:48 Blacktower kernel: [  111.401170]  drm_ioctl+0x234/0x3d0 [drm]
Oct 31 22:53:48 Blacktower kernel: [  111.401199]  ? drm_mode_getcrtc+0x190/0x190 [drm]
Oct 31 22:53:48 Blacktower kernel: [  111.401210]  ? __fsnotify_parent+0x9f/0x140
Oct 31 22:53:48 Blacktower kernel: [  111.401216]  do_vfs_ioctl+0x407/0x670
Oct 31 22:53:48 Blacktower kernel: [  111.401221]  ksys_ioctl+0x67/0x90
Oct 31 22:53:48 Blacktower kernel: [  111.401226]  __x64_sys_ioctl+0x1a/0x20
Oct 31 22:53:48 Blacktower kernel: [  111.401234]  do_syscall_64+0x57/0x190
Oct 31 22:53:48 Blacktower kernel: [  111.401242]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
Oct 31 22:53:48 Blacktower kernel: [  111.401246] RIP: 0033:0x7f592405b50b
Oct 31 22:53:48 Blacktower kernel: [  111.401251] Code: 0f 1e fa 48 8b 05 85 39 0d 00 64 c7 00 26 00 00 00 48 c7 c0 ff ff ff ff c3 66 0f 1f 44 00 00 f3 0f 1e fa b8 10 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 55 39 0d 00 f7 d8 64 89 01 48
Oct 31 22:53:48 Blacktower kernel: [  111.401255] RSP: 002b:00007ffc456e4e28 EFLAGS: 00003246 ORIG_RAX: 0000000000000010
Oct 31 22:53:48 Blacktower kernel: [  111.401259] RAX: ffffffffffffffda RBX: 00007ffc456e4e60 RCX: 00007f592405b50b
Oct 31 22:53:48 Blacktower kernel: [  111.401262] RDX: 00007ffc456e4e60 RSI: 00000000c06864a2 RDI: 0000000000000044
Oct 31 22:53:48 Blacktower kernel: [  111.401265] RBP: 00000000c06864a2 R08: 0000000000000000 R09: 00005602ea379310
[... cut ...]
Oct 31 22:53:48 Blacktower kernel: [  111.401392] FS:  00007f5923cfba80(0000) GS:ffff8f18deb00000(0000) knlGS:0000000000000000
Oct 31 22:53:48 Blacktower kernel: [  111.401395] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Oct 31 22:53:48 Blacktower kernel: [  111.401398] CR2: ffffafaec7130004 CR3: 0000000874070005 CR4: 00000000003606e0
Oct 31 22:53:48 Blacktower kernel: [  111.401401] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
Oct 31 22:53:48 Blacktower kernel: [  111.401404] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400

klogg, 2020-10-31:
I would like to suggest moving failing 2nd part (crash after sleep) out of this this particular issue: suspend-resume is not implemented in the driver yet

klogg, 2020-10-31:
Issue solved with removed IOREMAP flag in urb pages vmapping

closed 2020-10-31 by klogg