klogg/fl2000_drm #35: insmod: ERROR: could not insert module fl2000.ko: Unknown symbol in module

meslin8752, 2021-01-07:
Hi!
After cloning and making, I tried to insmod the driver, but I got the following error:
```
$ sudo insmod fl2000.ko 
insmod: ERROR: could not insert module fl2000.ko: Unknown symbol in module
```
Inspecting using dmesg, I got the following messages:
```
$ dmesg
[ 1166.265137] fl2000: Unknown symbol drm_kms_helper_poll_fini (err -2)
[ 1166.265294] fl2000: Unknown symbol drm_kms_helper_poll_init (err -2)
[ 1166.265326] fl2000: Unknown symbol drm_plane_enable_fb_damage_clips (err -2)
[ 1166.265356] fl2000: Unknown symbol drm_fb_helper_lastclose (err -2)
[ 1166.265447] fl2000: Unknown symbol drm_atomic_helper_shutdown (err -2)
[ 1166.265480] fl2000: Unknown symbol drm_gem_fb_simple_display_pipe_prepare_fb (err -2)
[ 1166.265723] fl2000: Unknown symbol drm_atomic_helper_commit (err -2)
[ 1166.265753] fl2000: Unknown symbol drm_atomic_helper_check (err -2)
[ 1166.265944] fl2000: Unknown symbol drm_atomic_helper_damage_merged (err -2)
[ 1166.265999] fl2000: Unknown symbol drm_gem_fb_create_with_dirty (err -2)
[ 1166.266140] fl2000: Unknown symbol drm_gem_fb_get_obj (err -2)
[ 1166.266270] fl2000: Unknown symbol drm_fbdev_generic_setup (err -2)
[ 1166.266304] fl2000: Unknown symbol drm_simple_display_pipe_init (err -2)
[ 1166.266396] fl2000: Unknown symbol drm_kms_helper_hotplug_event (err -2)
```
I'm using Ubuntu 20.04, GCC version 9.3.0, and my adapter ID is 1d5c:2000.

Can you help me, please?
Thanks!

klogg, 2021-03-24:
I think you need to insmod drm modules prior to using this driver. can you please check script `scripts/startup.sh`? This will solve your problem

DMaxter, 2024-04-14:
I'm hitting the same error, however only 3 symbols are not loaded: `drm_fb_dma_get_gem_obj`, `drm_gem_dma_dumb_create` and `drm_gem_dma_prime_import_sg_table_vmap`. The `drm` module is not getting loaded, but my guess is because I'm using `nvidia-drm` instead

DMaxter, 2024-04-14:
I was able to solve it by enabling module `drm_dma_helper`

bredigian, 2024-05-21:
> I was able to solve it by enabling module `drm_dma_helper`

Hi, I've the same error and I cannot solve that. How did you enable module drm_dma_helper? Thanks.

bredigian, 2024-05-21:
> I was able to solve it by enabling module `drm_dma_helper`

Hi, how are you? I have a question. Were you able to load the kernel correctly after enabling that module? Because, I did that but it didn't work for me. What happened was that it tried to load it, the screen flashed, but it gave me an error in the terminal. And when I connected HDMI nothing happened. When trying to restart it gave an error like it couldn't restart and I had to do a forced shutdown. I have an Acer laptop with Intel integrated graphics and Nvidia dedicated graphics, and the linux kernel loaded is 6.5.0-35-generic.

DMaxter, 2024-05-21:
It worked fine for me, but still didn't manage to use my dock with this driver.

When I enabled `drm_dma_helper`, it just enabled and did nothing more

drcid98, 2024-10-04:
Hi @DMaxter, how were you able to enable `drm_dma_helper` ? can you describe the process you followed?

DMaxter, 2024-10-07:
Since I am on Arch, I added the module in `/etc/mkinitcpio.conf`, issued a reconfiguration with mkinitcpio and rebooted. After that module was enabled

klogg, 2026-10-27:
The original missing-symbol report concerns module dependencies: insmod does not load dependencies automatically. For installed modules, use depmod followed by modprobe; the current QEMU startup tooling also loads the required dependency modules.

The later drm_dma_helper reports concern the former DMA GEM implementation. Current master uses DRM shmem GEM helpers instead, and module build/load tests pass on the Linux 7.0 target.

Current build, packaging and loading instructions: https://github.com/klogg/fl2000_drm#build

The historical kernels reported here are not covered by the current Linux 7.0 support target. If dependency loading fails with a current build, please open a new issue with the exact revision, kernel configuration, loading commands and full kernel log.

closed 2026-10-27 by klogg