klogg/fl2000_drm #36: IT66121

ricodol, 2021-01-12:
First TKS for your work.
Compilation worked
insmod for both module worked
doned on debian kernel 5.8

but after connection 

 Bus 002 Device 005: ID 1d5c:2000 Fresco Logic FL2000/FL2000DX VGA/DVI/HDMI Adapter

there was an error : 

CPU: 2 PID: 5968 Comm: kworker/2:1 Tainted: G        WCIOE     5.8.0-0.bpo.2-amd64 #1 Debian 5.8.10-1~bpo10+1
[ 1765.944322] Hardware name: Dell Inc. XPS 13 9350/09JHRY, BIOS 1.12.2 12/15/2019
[ 1765.944368] Workqueue: usb_hub_wq hub_event [usbcore]
[ 1765.944377] RIP: 0010:__flush_work+0x1c3/0x1e0
[ 1765.944383] Code: eb e6 4c 8b 6b 20 e9 e6 fe ff ff 8b 4d 00 48 8b 55 08 83 e1 08 48 0f ba 6d 00 03 80 c9 f0 e9 46 ff ff ff 0f 0b e9 65 ff ff ff <0f> 0b 45 31 f6 e9 5b ff ff ff e8 ee 20 75 00 66 66 2e 0f 1f 84 00
[ 1765.944387] RSP: 0018:ffffa79f84297b40 EFLAGS: 00010246
[ 1765.944392] RAX: 0000000000000000 RBX: ffff953403d04f90 RCX: 0000000000000000
[ 1765.944396] RDX: 0000000000000001 RSI: 0000000000000001 RDI: ffff953403d04f90
[ 1765.944399] RBP: ffff953403d04f90 R08: ffffffffc03af228 R09: 0000000000000000
[ 1765.944401] R10: 0000000000000000 R11: ffffffff8ee50840 R12: 0000000000000000
[ 1765.944404] R13: ffffa79f84297be0 R14: 0000000000000001 R15: ffff9532b8821430
[ 1765.944408] FS:  0000000000000000(0000) GS:ffff95342f300000(0000) knlGS:0000000000000000
[ 1765.944412] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 1765.944416] CR2: 00007fb049ffd000 CR3: 000000010e00a005 CR4: 00000000003606e0
[ 1765.944419] Call Trace:
[ 1765.944435]  ? finish_wait+0x3c/0x80
[ 1765.944480]  ? usb_kill_urb+0xac/0xc0 [usbcore]
[ 1765.944487]  ? work_busy+0x80/0x80
[ 1765.944493]  __cancel_work_timer+0x10a/0x190
[ 1765.944502]  ? _cond_resched+0x15/0x30
[ 1765.944546]  ? usb_poison_urb+0x4c/0xb0 [usbcore]
[ 1765.944563]  fl2000_intr_destroy+0x41/0x90 [fl2000]
[ 1765.944613]  usb_unbind_interface+0x76/0x260 [usbcore]
[ 1765.944629]  device_release_driver_internal+0xe5/0x1c0
[ 1765.944639]  bus_remove_device+0xdb/0x140
[ 1765.944646]  device_del+0x171/0x3f0
[ 1765.944691]  ? usb_remove_ep_devs+0x1b/0x30 [usbcore]
[ 1765.944734]  usb_disable_device+0x8c/0x1a0 [usbcore]
[ 1765.944760]  usb_disconnect+0xbc/0x260 [usbcore]
[ 1765.944798]  hub_event+0xa18/0x1580 [usbcore]
[ 1765.944815]  ? __switch_to+0x7a/0x3b0
[ 1765.944823]  process_one_work+0x1ad/0x370
[ 1765.944832]  ? create_worker+0x1a0/0x1a0
[ 1765.944837]  worker_thread+0x30/0x390
[ 1765.944843]  ? create_worker+0x1a0/0x1a0
[ 1765.944850]  kthread+0x116/0x130
[ 1765.944858]  ? kthread_park+0x80/0x80
[ 1765.944866]  ret_from_fork+0x22/0x30
[ 1765.944874] ---[ end trace 3e5c0c57e7060676 ]---
[ 1849.686732] usb 2-1: new SuperSpeed Gen 1 USB device number 6 using xhci_hcd
[ 1849.707177] usb 2-1: config 1 interface 1 altsetting 0 has a duplicate endpoint with address 0x81, skipping
[ 1849.707182] usb 2-1: config 1 interface 1 altsetting 0 has a duplicate endpoint with address 0x1, skipping
[ 1849.707185] usb 2-1: config 1 interface 1 altsetting 1 has a duplicate endpoint with address 0x81, skipping
[ 1849.707189] usb 2-1: config 1 interface 1 altsetting 1 has a duplicate endpoint with address 0x1, skipping
[ 1849.707193] usb 2-1: config 1 interface 1 altsetting 2 has a duplicate endpoint with address 0x81, skipping
[ 1849.707196] usb 2-1: config 1 interface 1 altsetting 2 has a duplicate endpoint with address 0x1, skipping
[ 1849.707200] usb 2-1: config 1 interface 1 altsetting 3 has a duplicate endpoint with address 0x81, skipping
[ 1849.707203] usb 2-1: config 1 interface 1 altsetting 3 has a duplicate endpoint with address 0x1, skipping
[ 1849.707206] usb 2-1: config 1 interface 1 altsetting 4 has a duplicate endpoint with address 0x81, skipping
[ 1849.707209] usb 2-1: config 1 interface 1 altsetting 4 has a duplicate endpoint with address 0x1, skipping
[ 1849.707212] usb 2-1: config 1 interface 1 altsetting 5 has a duplicate endpoint with address 0x81, skipping
[ 1849.707215] usb 2-1: config 1 interface 1 altsetting 5 has a duplicate endpoint with address 0x1, skipping
[ 1849.707219] usb 2-1: config 1 interface 1 altsetting 6 has a duplicate endpoint with address 0x81, skipping
[ 1849.707222] usb 2-1: config 1 interface 1 altsetting 6 has a duplicate endpoint with address 0x1, skipping
[ 1849.707232] usb 2-1: New USB device found, idVendor=1d5c, idProduct=2000, bcdDevice= 2.00
[ 1849.707236] usb 2-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[ 1849.735667] usb 2-1: Configured FL2000 registers
[ 1849.735808] i2c i2c-9: Detecting IT66121 at address 0x4C on usb-0000:00:14.0-1
[ 1849.739283] i2c i2c-9: I2C error detected: status 15
[... 36 lines cut ...]
[ 1765.944480]  ? usb_kill_urb+0xac/0xc0 [usbcore]
[ 1765.944487]  ? work_busy+0x80/0x80
[ 1765.944493]  __cancel_work_timer+0x10a/0x190
[ 1765.944502]  ? _cond_resched+0x15/0x30
[ 1765.944546]  ? usb_poison_urb+0x4c/0xb0 [usbcore]
[ 1765.944563]  fl2000_intr_destroy+0x41/0x90 [fl2000]
[ 1765.944613]  usb_unbind_interface+0x76/0x260 [usbcore]
[ 1765.944629]  device_release_driver_internal+0xe5/0x1c0
[ 1765.944639]  bus_remove_device+0xdb/0x140
[ 1765.944646]  device_del+0x171/0x3f0
[ 1765.944691]  ? usb_remove_ep_devs+0x1b/0x30 [usbcore]
[ 1765.944734]  usb_disable_device+0x8c/0x1a0 [usbcore]
[ 1765.944760]  usb_disconnect+0xbc/0x260 [usbcore]
[ 1765.944798]  hub_event+0xa18/0x1580 [usbcore]
[ 1765.944815]  ? __switch_to+0x7a/0x3b0
[ 1765.944823]  process_one_work+0x1ad/0x370
[ 1765.944832]  ? create_worker+0x1a0/0x1a0
[ 1765.944837]  worker_thread+0x30/0x390
[ 1765.944843]  ? create_worker+0x1a0/0x1a0
[ 1765.944850]  kthread+0x116/0x130
[ 1765.944858]  ? kthread_park+0x80/0x80
[ 1765.944866]  ret_from_fork+0x22/0x30
[ 1765.944874] ---[ end trace 3e5c0c57e7060676 ]---

if someone has some clues…

klogg, 2021-03-24:
Please try with latest commit. I did some rework of how it66121 / fl2000 register and interact that allowed me to handle deregistration better and fixed some races. I am not sure everything is fixed now but worth to try

Boteium, 2021-04-01:
I believe I have similar problem on a 5.10 kernel.
the module compiled and loaded sucessfully.
xrandr does shows new HDMI display but says "unknown connection".

[421796.887484] fl2000: loading out-of-tree module taints kernel.
[421796.887527] fl2000: module verification failed: signature and/or required key missing - tainting kernel
[421796.889030] usbcore: registered new interface driver fl2000_usb
[421818.707970] usb 4-4: new SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[421818.728254] usb 4-4: config 1 interface 1 altsetting 0 has a duplicate endpoint with address 0x81, skipping
[421818.728256] usb 4-4: config 1 interface 1 altsetting 0 has a duplicate endpoint with address 0x1, skipping
[421818.728257] usb 4-4: config 1 interface 1 altsetting 1 has a duplicate endpoint with address 0x81, skipping
[421818.728259] usb 4-4: config 1 interface 1 altsetting 1 has a duplicate endpoint with address 0x1, skipping
[421818.728262] usb 4-4: config 1 interface 1 altsetting 2 has a duplicate endpoint with address 0x81, skipping
[421818.728263] usb 4-4: config 1 interface 1 altsetting 2 has a duplicate endpoint with address 0x1, skipping
[421818.728265] usb 4-4: config 1 interface 1 altsetting 3 has a duplicate endpoint with address 0x81, skipping
[421818.728266] usb 4-4: config 1 interface 1 altsetting 3 has a duplicate endpoint with address 0x1, skipping
[421818.728267] usb 4-4: config 1 interface 1 altsetting 4 has a duplicate endpoint with address 0x81, skipping
[421818.728268] usb 4-4: config 1 interface 1 altsetting 4 has a duplicate endpoint with address 0x1, skipping
[421818.728269] usb 4-4: config 1 interface 1 altsetting 5 has a duplicate endpoint with address 0x81, skipping
[421818.728270] usb 4-4: config 1 interface 1 altsetting 5 has a duplicate endpoint with address 0x1, skipping
[421818.728271] usb 4-4: config 1 interface 1 altsetting 6 has a duplicate endpoint with address 0x81, skipping
[421818.728272] usb 4-4: config 1 interface 1 altsetting 6 has a duplicate endpoint with address 0x1, skipping
[421818.728275] usb 4-4: New USB device found, idVendor=1d5c, idProduct=2000, bcdDevice= 2.00
[421818.728276] usb 4-4: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[421818.735039] i2c i2c-7: Binding FL2000 master
[421818.735356] i2c 0-004c: Binding IT66121 component
[421818.889040] i2c i2c-7: Bridge attached
[421818.889048] i2c i2c-7: bound 0-004c (ops it66121_component_ops [it66121])
[421818.889172] [drm] Initialized fl2000_drm 0.0.1 20181001 for i2c-7 on minor 1
[421818.910774] i2c i2c-7: Cannot get monitor status (-6)
[421818.911331] i2c i2c-7: [drm] Cannot find any crtc or sizes
[421818.925185] i2c i2c-7: Cannot get monitor status (-6)
[421818.926326] i2c i2c-7: Cannot get monitor status (-6)
[421819.008500] i2c i2c-7: Cannot read interrupt status (-6)
[421819.163534] i2c i2c-7: Cannot get monitor status (-6)
[421819.164094] i2c i2c-7: [drm] Cannot find any crtc or sizes
[421819.164642] i2c i2c-7: Cannot get monitor status (-6)
[421819.165752] i2c i2c-7: Cannot get monitor status (-6)
[421826.631923] i2c i2c-7: Cannot get monitor status (-6)
[421826.633271] i2c i2c-7: [drm] Cannot find any crtc or sizes
[421826.633835] i2c i2c-7: Cannot get monitor status (-6)
[421826.634953] i2c i2c-7: Cannot get monitor status (-6)
[421835.190513] i2c i2c-7: Cannot get monitor status (-6)
[421837.895914] i2c i2c-7: Cannot get monitor status (-6)
[421842.115557] i2c i2c-7: Cannot get monitor status (-6)
[421842.116120] i2c i2c-7: [drm] Cannot find any crtc or sizes
[421842.117249] i2c i2c-7: Cannot get monitor status (-6)
[421842.118365] i2c i2c-7: Cannot get monitor status (-6)
[421842.391992] i2c i2c-7: Cannot get monitor status (-6)
[421842.392567] i2c i2c-7: [drm] Cannot find any crtc or sizes
[421842.393165] i2c i2c-7: Cannot get monitor status (-6)
[421842.394290] i2c i2c-7: Cannot get monitor status (-6)
[421913.289838] i2c i2c-7: Cannot get monitor status (-6)
[421914.209711] i2c i2c-7: Cannot get monitor status (-6)
[421945.148711] ------------[ cut here ]------------
[421945.148716] WARNING: CPU: 8 PID: 1128 at kernel/dma/mapping.c:188 dma_map_sg_attrs+0x37/0x50
[421945.148717] Modules linked in: it66121(OE) fl2000(OE) snd_seq_dummy snd_hrtimer snd_seq rfcomm xt_MASQUERADE ip6table_mangle iptable_mangle iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 iptable_filter ip6table_filter ip6_tables ext4 mbcache jbd2 i2c_dev cmac algif_hash algif_skcipher af_alg tcp_bbr sch_fq bnep hwmon_vid nls_iso8859_1 vfat fat amdgpu snd_hda_codec_realtek snd_hda_codec_generic ledtrig_audio eeepc_wmi uvcvideo gpu_sched asus_wmi sparse_keymap i2c_algo_bit video wmi_bmof snd_usb_audio snd_hda_intel videobuf2_vmalloc ttm snd_intel_dspcfg videobuf2_memops soundwire_intel amd64_edac_mod videobuf2_v4l2 edac_mce_amd soundwire_generic_allocation videobuf2_common soundwire_cadence kvm_amd snd_usbmidi_lib snd_hda_codec drm_kms_helper snd_rawmidi videodev snd_seq_device kvm snd_hda_core snd_hwdep soundwire_bus btusb snd_soc_core cec btrtl syscopyarea btbcm sysfillrect btintel snd_compress ac97_bus snd_pcm_dmaengine sp5100_tco sysimgblt irqbypass rapl bluetooth k10temp
[421945.148750]  i2c_piix4 snd_pcm mc fb_sys_fops snd_timer snd joydev mousedev soundcore ecdh_generic uas rfkill ecc crc16 wmi gpio_amdpt gpio_generic mac_hid pinctrl_amd acpi_cpufreq drm pkcs8_key_parser sg fuse crypto_user agpgart ip_tables x_tables btrfs blake2b_generic libcrc32c crc32c_generic xor raid6_pq dm_crypt cbc encrypted_k [... cut ...]
[421945.148907] RSP: 002b:00007ffeebdd1e38 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
[421945.148907] RAX: ffffffffffffffda RBX: 00007ffeebdd1e7c RCX: 00007fd2e7be7e6b
[421945.148908] RDX: 00007ffeebdd1e7c RSI: 00000000c00c642e RDI: 000000000000003d
[421945.148908] RBP: 00000000c00c642e R08: 000000000000003d R09: 0000000000000000
[421945.148909] R10: fffffffffffffbaf R11: 0000000000000246 R12: 0000000000000000
[421945.148909] R13: 000000000000003d R14: 0000000000000320 R15: 0000000000001400
[421945.148910] ---[ end trace b1d97ece89f5589e ]---

klogg, 2021-06-25:
hi, thank you for the logs, checking

klogg, 2026-10-27:
Closing this historical report as obsolete following substantial changes to the driver. Current master has reworked USB teardown and DRM lifetime handling, replaced the custom GEM implementation with shmem helpers, and rebased the bridge onto the upstream IT66121 driver with transport adaptations.

EDID, visible HDMI output, hotplug and lifecycle tests pass on the available hardware with Linux 7.0. This does not establish that every adapter or failure discussed in this thread is fixed.

If the same problem persists with current master on the documented Linux 7.0 target, please reopen this issue with the exact driver revision, adapter/output type, USB topology, reproduction steps, full kernel log and enumerator service log. Please open a separate issue for a different failure. Native VGA support is tracked in #79.

Current setup: https://github.com/klogg/fl2000_drm#build
Validation scope: https://github.com/klogg/fl2000_drm/blob/master/docs/DEVELOPMENT.md#validation-scope

closed 2026-10-27 by klogg