klogg/fl2000_drm #37: Unable to compile on Kernel 5.10.5

aroooo, 2021-01-15:
Hi, first of all I wanted to thank you for making this driver.
However, the driver is unable to be compiled and used on my machine.
Here's my configuration and my build log:

[https://pastebin.com/YHv3vuKL](make log) 
OS: EndeavousOS (Arch)
Kernel: 5.10.5-arch1-1

Thank you.

Fachpersonal, 2021-01-23:
Same here.
OS: Manjaro 20.2.1 (Arch)
Kernel: 5.10.7-3-MANJARO

`~/tmp/fl2000_drm >>> make                                                                                                                                              [130]
make CHECK="/usr/bin/sparse" -C /lib/modules/5.10.7-3-MANJARO/build M=/home/fachpersonal/tmp/fl2000_drm modules
make[1]: Entering directory '/usr/lib/modules/5.10.7-3-MANJARO/build'
  CC [M]  /home/fachpersonal/tmp/fl2000_drm/fl2000_gem.o
/home/fachpersonal/tmp/fl2000_drm/fl2000_gem.c: In function ‘fl2000_gem_create_with_handle’:
/home/fachpersonal/tmp/fl2000_drm/fl2000_gem.c:128:2: error: implicit declaration of function ‘drm_gem_object_put_unlocked’; did you mean ‘drm_gem_object_put_locked’? [-Werror=implicit-function-declaration]
  128 |  drm_gem_object_put_unlocked(gem_obj);
      |  ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |  drm_gem_object_put_locked
/home/fachpersonal/tmp/fl2000_drm/fl2000_gem.c: In function ‘fl2000_gem_prime_get_sg_table’:
/home/fachpersonal/tmp/fl2000_drm/fl2000_gem.c:204:34: error: passing argument 1 of ‘drm_prime_pages_to_sg’ from incompatible pointer type [-Werror=incompatible-pointer-types]
  204 |  return drm_prime_pages_to_sg(obj->pages, obj->num_pages);
      |                               ~~~^~~~~~~
      |                                  |
      |                                  struct page **
In file included from /home/fachpersonal/tmp/fl2000_drm/fl2000.h:27,
                 from /home/fachpersonal/tmp/fl2000_drm/fl2000_gem.c:14:
./include/drm/drm_prime.h:91:59: note: expected ‘struct drm_device *’ but argument is of type ‘struct page **’
   91 | struct sg_table *drm_prime_pages_to_sg(struct drm_device *dev,
      |                                        ~~~~~~~~~~~~~~~~~~~^~~
/home/fachpersonal/tmp/fl2000_drm/fl2000_gem.c:204:46: warning: passing argument 2 of ‘drm_prime_pages_to_sg’ makes pointer from integer without a cast [-Wint-conversion]
  204 |  return drm_prime_pages_to_sg(obj->pages, obj->num_pages);
      |                                           ~~~^~~~~~~~~~~
      |                                              |
      |                                              size_t {aka long unsigned int}
In file included from /home/fachpersonal/tmp/fl2000_drm/fl2000.h:27,
                 from /home/fachpersonal/tmp/fl2000_drm/fl2000_gem.c:14:
./include/drm/drm_prime.h:92:26: note: expected ‘struct page **’ but argument is of type ‘size_t’ {aka ‘long unsigned int’}
   92 |            struct page **pages, unsigned int nr_pages);
      |            ~~~~~~~~~~~~~~^~~~~
/home/fachpersonal/tmp/fl2000_drm/fl2000_gem.c:204:9: error: too few arguments to function ‘drm_prime_pages_to_sg’
  204 |  return drm_prime_pages_to_sg(obj->pages, obj->num_pages);
      |         ^~~~~~~~~~~~~~~~~~~~~
In file included from /home/fachpersonal/tmp/fl2000_drm/fl2000.h:27,
                 from /home/fachpersonal/tmp/fl2000_drm/fl2000_gem.c:14:
./include/drm/drm_prime.h:91:18: note: declared here
   91 | struct sg_table *drm_prime_pages_to_sg(struct drm_device *dev,
      |                  ^~~~~~~~~~~~~~~~~~~~~
/home/fachpersonal/tmp/fl2000_drm/fl2000_gem.c:205:1: error: control reaches end of non-void function [-Werror=return-type]
  205 | }
      | ^
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:279: /home/fachpersonal/tmp/fl2000_drm/fl2000_gem.o] Error 1
make[1]: *** [Makefile:1805: /home/fachpersonal/tmp/fl2000_drm] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.10.7-3-MANJARO/build'
make: *** [Makefile:21: modules] Error 2
`

dkuku, 2021-02-11:
on arch for me it helped to change all the -Werror-xxxxxx to --Wno-error-xxxxx in `/lib/modules/5.10.7-3-MANJARO/build/Makefile`
but now it fails on:
    fl2000_drm/fl2000_gem.c:206:9: error: too few arguments to function ‘drm_prime_pages_to_sg’
this is because of some change in kernel 5.10

guzzard, 2021-02-18:
I got the driver to build on 5.10.16 with below changes. This will probably not produce a working driver since I'm passing NULL and commented out parts. I had no success in getting monitor detected, however driver detected the USB-to-VGA-dongle and I could see I2C getting detected etc.

I did this mostly to see how big the issues are that need solving. Seems like there at least three things that need to be addressed.

1. The definition ```#define drm_gem_object_put_unlocked drm_gem_object_put``` has been removed from the kernel headers, so I added it in fl2000.h
Related commit: https://github.com/torvalds/linux/commit/ab15d56e27be47b7feebffd6e8319ece01959fbf#diff-e9bed6d53fbcd939e0599deb4615e47c0fc94faea85a03cdccc20db9ee8437d8
2. Function ```int drm_dev_init(struct drm_device *dev, struct drm_driver *driver, struct device *parent);``` had been removed from the kernel headers. I commented out the call to that function. I think it needs to be replaced with a function call to ```void *__devm_drm_dev_alloc(struct device *parent, struct drm_driver *driver, size_t size, size_t offset);``` instead?
Related commit: https://github.com/torvalds/linux/commit/a7d39439f8bd14b5becebc095aef3f1ec3191fe2#diff-f0d72b760bb952a829efa24fdb307d82aaf370ab67be0957640511d12f2e0752
3. Function ```struct sg_table *drm_prime_pages_to_sg(struct page **pages, unsigned int nr_pages);``` has been changed in the kernel headers to include a pointer to the drm_device: ```struct sg_table *drm_prime_pages_to_sg(struct drm_device *dev, struct page **pages, unsigned int nr_pages);```. I passed NULL instead of a pointer to drm_device.
Related commit: https://github.com/torvalds/linux/commit/6ea6be77086f23d4b346c9946dae24593befda2e#diff-ed8ce72408c8b6384bc6897361ad0b22907c580e48584ba334cae7e8e7f23071

Patch with the changes below. Hopefully @klogg or someone in the community with more knowledge can fix it.
```
diff -ruN fl2000_drm/fl2000_drm.c fl2000_drm_test/fl2000_drm.c
--- fl2000_drm/fl2000_drm.c	2021-02-18 22:31:11.415476585 +0100
+++ fl2000_drm_test/fl2000_drm.c	2021-02-18 22:09:26.389079771 +0100
@@ -546,11 +546,11 @@
 	devres_add(master, drm_if);
 
 	drm = &drm_if->drm;
-	ret = drm_dev_init(drm, &fl2000_drm_driver, master);
-	if (ret) {
-		dev_err(master, "Cannot initialize DRM device (%d)", ret);
-		return ret;
-	}
+	// ret = drm_dev_init(drm, &fl2000_drm_driver, master);
+	// if (ret) {
+	// 	dev_err(master, "Cannot initialize DRM device (%d)", ret);
+	// 	return ret;
+	// }
 
 	/* For register operations */
 	drm->dev_private = usb_dev;
diff -ruN fl2000_drm/fl2000_gem.c fl2000_drm_test/fl2000_gem.c
--- fl2000_drm/fl2000_gem.c	2021-02-18 22:31:11.415476585 +0100
+++ fl2000_drm_test/fl2000_gem.c	2021-02-18 21:53:20.643330719 +0100
@@ -201,7 +201,7 @@
 	if (!obj->pages)
 		return ERR_PTR(-ENOMEM);
 
-	return drm_prime_pages_to_sg(obj->pages, obj->num_pages);
+    return drm_prime_pages_to_sg(NULL, obj->pages, obj->num_pages);
 }
 
 struct drm_gem_object *fl2000_gem_prime_import_sg_table(struct drm_device *drm,
diff -ruN fl2000_drm/fl2000.h fl2000_drm_test/fl2000.h
--- fl2000_drm/fl2000.h	2021-02-18 22:31:11.415476585 +0100
+++ fl2000_drm_test/fl2000.h	2021-02-18 21:53:48.873401865 +0100
@@ -41,6 +41,8 @@
 
 #if defined(CONFIG_DEBUG_FS)
 
+#define drm_gem_object_put_unlocked drm_gem_object_put
+
 /* NOTE: it might be unsafe / insecure to use allow rw access everyone */
 #include <linux/debugfs.h>
 static const umode_t fl2000_debug_umode = 0666;
```

raitech, 2021-02-19:
Hello folks!

I will fiddle with this issue in the next couple of days. Let's see what I can do ;)

raitech, 2021-02-21:
Hi there!

I have made a PR with some modifications that let it compile on 5.10 kernel.

https://github.com/klogg/fl2000_drm/pull/38

Enjoy :)

aroooo, 2021-03-05:
> Hi there!
> 
> I have made a PR with some modifications that let it compile on 5.10 kernel.
> 
> #38
> 
> Enjoy :)

Thanks, but there remain a little problem:
I've recently updated to Kernel 5.11.2 and encountered 2 new errors I can't fix, not knowing C++, here's the log:
[make log](https://pastebin.com/APswnYGn)
Notify me if you find/make a fix, thank you and have a great day!

guzzard, 2021-03-06:
@aroooo try this branch in raitechs repo: https://github.com/raitech/fl2000_drm/tree/solving-issue-37-to-compile-on-5.11

aroooo, 2021-03-07:
> @aroooo try this branch in raitechs repo: https://github.com/raitech/fl2000_drm/tree/solving-issue-37-to-compile-on-5.11

IT COMPILES! :pray: 
Thank you.

closed 2021-03-07 by aroooo

klogg, 2021-03-24:
Please see 5.10 support added in 19f1252