klogg/fl2000_drm #39: Getting black screen before insert module

cvivieca, 2021-03-09:
Before running insmod command I'm getting a black screen on both laptop screen and screen with the adapter. I just can see the cursor. 

No idea why.

cvivieca, 2021-03-21:
Just an update about this. I have been testing and I know why the problem happen. If you are getting this you have to close all app like Edge, Chrome, VS Code and run insmod. After doing this you can open yours apps without problem.

klogg, 2026-10-27:
Closing this historical Linux 5.4/5.8 report as obsolete. The driver has since undergone substantial DRM, framebuffer, bridge and USB lifecycle changes, and the current supported target is Linux 7.0. The application-closing workaround discussed above has not been established as necessary for current master.

If the same black-screen problem persists with current master on Linux 7.0, please reopen this issue with the exact driver revision, GPU/compositor or desktop details, adapter/output type, reproduction steps and full kernel log. Please clarify whether the failure occurs before loading the module, during loading, or when the USB adapter is connected. For a different failure, please open a separate issue.

Current setup: https://github.com/klogg/fl2000_drm#build
Validation scope: https://github.com/klogg/fl2000_drm/blob/master/docs/DEVELOPMENT.md#validation-scope

Closure marks the old report as obsolete; it does not claim that this specific desktop interaction has been reproduced and fixed.

closed 2026-10-27 by klogg