klogg/fl2000_drm #41: not working on Manjaro, kernel 5.10.30

eduardosm, 2021-04-21:
I am trying to test the driver on Manjaro with kernel 5.10.30. I have successfully built the module and loaded the `drm`, `drm_kms_helper` and `fl2000` modules. I have also added `quirks=1D5C:2000:USB_QUIRK_NO_LPM` to boot params. However, after plugging the dongle, nothing apparently happens.

lsusb lists the device as:
```
Bus 002 Device 008: ID 1d5c:2000 Fresco Logic FL2000/FL2000DX VGA/DVI/HDMI Adapter
```
```
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/8p, 10000M
    |__ Port 1: Dev 8, If 0, Class=, Driver=fl2000_usb, 5000M
    |__ Port 1: Dev 8, If 1, Class=, Driver=fl2000_usb, 5000M
    |__ Port 1: Dev 8, If 2, Class=, Driver=fl2000_usb, 5000M
    |__ Port 1: Dev 8, If 3, Class=Mass Storage, Driver=usb-storage, 5000M
```

And dmesg shows:
```
[ 1132.833822] usb 1-1: new high-speed USB device number 17 using xhci_hcd
[ 1133.014164] usb 1-1: New USB device found, idVendor=1a40, idProduct=0801, bcdDevice= 1.00
[ 1133.014171] usb 1-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[ 1133.014175] usb 1-1: Product: USB 2.0 Hub
[ 1133.015176] hub 1-1:1.0: USB hub found
[ 1133.015218] hub 1-1:1.0: 4 ports detected
[ 1133.173527] usb 2-1: new SuperSpeed Gen 1 USB device number 8 using xhci_hcd
[ 1133.203811] usb 2-1: config 1 interface 1 altsetting 0 has a duplicate endpoint with address 0x81, skipping
[ 1133.203813] usb 2-1: config 1 interface 1 altsetting 0 has a duplicate endpoint with address 0x1, skipping
[ 1133.203813] usb 2-1: config 1 interface 1 altsetting 1 has a duplicate endpoint with address 0x81, skipping
[ 1133.203814] usb 2-1: config 1 interface 1 altsetting 1 has a duplicate endpoint with address 0x1, skipping
[ 1133.203815] usb 2-1: config 1 interface 1 altsetting 2 has a duplicate endpoint with address 0x81, skipping
[ 1133.203816] usb 2-1: config 1 interface 1 altsetting 2 has a duplicate endpoint with address 0x1, skipping
[ 1133.203816] usb 2-1: config 1 interface 1 altsetting 3 has a duplicate endpoint with address 0x81, skipping
[ 1133.203817] usb 2-1: config 1 interface 1 altsetting 3 has a duplicate endpoint with address 0x1, skipping
[ 1133.203818] usb 2-1: config 1 interface 1 altsetting 4 has a duplicate endpoint with address 0x81, skipping
[ 1133.203818] usb 2-1: config 1 interface 1 altsetting 4 has a duplicate endpoint with address 0x1, skipping
[ 1133.203819] usb 2-1: config 1 interface 1 altsetting 5 has a duplicate endpoint with address 0x81, skipping
[ 1133.203820] usb 2-1: config 1 interface 1 altsetting 5 has a duplicate endpoint with address 0x1, skipping
[ 1133.203821] usb 2-1: config 1 interface 1 altsetting 6 has a duplicate endpoint with address 0x81, skipping
[ 1133.203821] usb 2-1: config 1 interface 1 altsetting 6 has a duplicate endpoint with address 0x1, skipping
[ 1133.203824] usb 2-1: New USB device found, idVendor=1d5c, idProduct=2000, bcdDevice= 2.00
[ 1133.203825] usb 2-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[ 1133.205429] usb-storage 2-1:1.3: USB Mass Storage device detected
[ 1133.205639] scsi host5: usb-storage 2-1:1.3º
[ 1133.343880] usb 1-1.2: new full-speed USB device number 18 using xhci_hcd
[ 1133.480989] usb 1-1.2: New USB device found, idVendor=0d8c, idProduct=013c, bcdDevice= 1.00
[ 1133.480996] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1133.481000] usb 1-1.2: Product: USB PnP Sound Device
[ 1133.481003] usb 1-1.2: Manufacturer: C-Media Electronics Inc.      
[ 1133.499085] input: C-Media Electronics Inc.       USB PnP Sound Device as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.2/1-1.2:1.3/0003:0D8C:013C.0007/input/input30
[ 1133.563835] hid-generic 0003:0D8C:013C.0007: input,hidraw0: USB HID v1.00 Device [C-Media Electronics Inc.       USB PnP Sound Device] on usb-0000:00:14.0-1.2/input3
[ 1134.223821] scsi 5:0:0:0: Direct-Access                               A001 PQ: 0 ANSI: 4
[ 1134.224054] sd 5:0:0:0: Attached scsi generic sg1 type 0
[ 1134.224487] sd 5:0:0:0: [sdb] 2047 4096-byte logical blocks: (8.38 MB/8.00 MiB)
[ 1134.224621] sd 5:0:0:0: [sdb] Write Protect is off
[ 1134.224622] sd 5:0:0:0: [sdb] Mode Sense: 03 00 00 00
[ 1134.224727] sd 5:0:0:0: [sdb] No Caching mode page found
[ 1134.224729] sd 5:0:0:0: [sdb] Assuming drive cache: write through
[ 1134.296277]  sdb:
[ 1134.297678] sd 5:0:0:0: [sdb] Attached SCSI disk
[ 1164.633903] usb 2-1: reset SuperSpeed Gen 1 USB device number 8 using xhci_hcd
```

Skippythekangoo, 2021-11-30:
What did you do for builting succefully.
I'm with Archolinux 5.14.16-arch1-1 and I have a same isssue like some people.
thank you.

klogg, 2026-10-27:
Closing this historical Linux 5.10 report as obsolete after substantial changes to bridge discovery, I2C handling, DRM setup and USB lifecycle management. The current supported target is Linux 7.0.

Binding the FL2000 USB interfaces alone does not establish that a usable display pipeline has attached. Current installation includes a udev-activated enumerator service that identifies and instantiates the external bridge on the adapter's own I2C bus.

Please follow the current setup instructions: https://github.com/klogg/fl2000_drm#bridge-configuration-on-x86

If the same failure persists with current master on Linux 7.0, please reopen this issue with the exact revision, adapter/output type, USB topology, full kernel log, enumerator service status/log and connector status/EDID where available. Native VGA remains tracked in #79. Closure does not claim that this particular adapter has been retested.

closed 2026-10-27 by klogg