klogg/fl2000_drm #42: Driver installed but display still not working

theaayushanand, 2021-04-26:
Hello, 

I have installed the driver according to the steps mentioned in the README but i still dont get my display to be working , are there any other steps required to get it all working ? 
Thanks in advance

GustavoEmmel, 2021-05-21:
Same in here.
Instaled on ubuntu 20.10 as README but still not working

klogg, 2021-06-25:
Please provide kernel logs (dmesg)

davidea72, 2022-05-20:
same problem, 
debian 11 with KDE
5.10.0-14-amd64 #1 SMP Debian 5.10.113-1 (2022-04-29) x86_64 GNU/Linux

916.215262] usb 1-1.2: USB disconnect, device number 9
[  919.269318] usb 1-1: new high-speed USB device number 10 using xhci_hcd
[  919.421432] usb 1-1: New USB device found, idVendor=1a40, idProduct=0101, bcdDevice= 1.00
[  919.421436] usb 1-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[  919.421438] usb 1-1: Product: USB2.0 HUB
[  919.422165] hub 1-1:1.0: USB hub found
[  919.422189] hub 1-1:1.0: 4 ports detected
[  919.809234] usb 1-1.2: new full-speed USB device number 11 using xhci_hcd
[  920.014946] usb 1-1.2: New USB device found, idVendor=2000, idProduct=dd00, bcdDevice=11.16
[  920.014950] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  920.014952] usb 1-1.2: Product: HZD100
[  920.014953] usb 1-1.2: Manufacturer: C&T
[  920.014955] usb 1-1.2: SerialNumber: U2dc0z7ws53a79rn
[  920.025523] input: C&T HZD100 as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.2/1-1.2:1.3/0003:2000:DD00.0004/input/input23
[  920.085485] hid-generic 0003:2000:DD00.0004: input,hidraw0: USB HID v1.00 Device [C&T HZD100] on usb-0000:00:14.0-1.2/input3
[  967.761575] INFO: task kworker/5:0:14307 blocked for more than 604 seconds.
[  967.761582]       Tainted: G           O      5.10.0-14-amd64 #1 Debian 5.10.113-1
[  967.761583] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  967.761586] task:kworker/5:0     state:D stack:    0 pid:14307 ppid:     2 flags:0x00004000
[  967.761632] Workqueue: usb_hub_wq hub_event [usbcore]
[  967.761634] Call Trace:
[  967.761649]  __schedule+0x282/0x870
[  967.761654]  ? usleep_range+0x80/0x80
[  967.761656]  schedule+0x46/0xb0
[  967.761659]  schedule_timeout+0xff/0x140
[  967.761669]  ? __prepare_to_swait+0x4b/0x70
[  967.761671]  __wait_for_common+0xae/0x160
[  967.761682]  i2c_del_adapter+0x259/0x310
[  967.761689]  release_nodes+0x225/0x270
[  967.761697]  __device_release_driver+0x18a/0x250
[  967.761700]  device_release_driver+0x24/0x30
[  967.761702]  bus_remove_device+0xdb/0x140
[  967.761704]  device_del+0x18b/0x430
[  967.761723]  usb_disconnect.cold+0x135/0x20a [usbcore]
[  967.761733]  hub_event+0xbbf/0x1850 [usbcore]
[  967.761738]  ? __update_idle_core+0x6d/0xa0
[  967.761742]  ? __switch_to_asm+0x42/0x70
[  967.761750]  ? __switch_to+0x114/0x450
[  967.761758]  process_one_work+0x1b6/0x350
[  967.761761]  worker_thread+0x53/0x3e0
[  967.761763]  ? process_one_work+0x350/0x350
[  967.761768]  kthread+0x11b/0x140
[  967.761771]  ? __kthread_bind_mask+0x60/0x60
[  967.761774]  ret_from_fork+0x22/0x30

jakre1234, 2023-06-27:
Did anyone got success here ??

larrasket, 2023-06-27:
@jakre1234 Nope. I recommend you to search for another chip unless you are a kernel developer. I personally bought a new laptop with multi-display adapter support. It's shame that FL developer won't just develop a driver for linux.

KoukiMatsuda, 2024-07-23:
I installed a driver on my Ubuntu 20.04 environment, but nothing is displayed on the screen. 
Below are the details of the operating environment and the logs during operation.
Does anyone know anything about this?

[Operating Environment]
OS: Ubuntu 20.04
Kernel: 5.8

SH-A: 37bc1df54c47833c1416f4edbcfe75f17a1a78b6

[Operation Logs]
■ USB Device Connection
```shell
$ dmesg
[ 2811.540146] usb 1-4: new high-speed USB device number 3 using xhci_hcd
[ 2811.688327] usb 1-4: New USB device found, idVendor=1a40, idProduct=0801, bcdDevice= 1.00
[ 2811.688331] usb 1-4: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[ 2811.688333] usb 1-4: Product: USB 2.0 Hub
[ 2811.689140] hub 1-4:1.0: USB hub found
[ 2811.689160] hub 1-4:1.0: 4 ports detected
[ 2811.816176] usb 2-4: new SuperSpeed Gen 1 USB device number 3 using xhci_hcd
[ 2811.836413] usb 2-4: config 1 interface 1 altsetting 0 has a duplicate endpoint with address 0x81, skipping
[ 2811.836417] usb 2-4: config 1 interface 1 altsetting 0 has a duplicate endpoint with address 0x1, skipping
[ 2811.836420] usb 2-4: config 1 interface 1 altsetting 1 has a duplicate endpoint with address 0x81, skipping
[ 2811.836423] usb 2-4: config 1 interface 1 altsetting 1 has a duplicate endpoint with address 0x1, skipping
[ 2811.836425] usb 2-4: config 1 interface 1 altsetting 2 has a duplicate endpoint with address 0x81, skipping
[ 2811.836428] usb 2-4: config 1 interface 1 altsetting 2 has a duplicate endpoint with address 0x1, skipping
[ 2811.836430] usb 2-4: config 1 interface 1 altsetting 3 has a duplicate endpoint with address 0x81, skipping
[ 2811.836432] usb 2-4: config 1 interface 1 altsetting 3 has a duplicate endpoint with address 0x1, skipping
[ 2811.836435] usb 2-4: config 1 interface 1 altsetting 4 has a duplicate endpoint with address 0x81, skipping
[ 2811.836437] usb 2-4: config 1 interface 1 altsetting 4 has a duplicate endpoint with address 0x1, skipping
[ 2811.836439] usb 2-4: config 1 interface 1 altsetting 5 has a duplicate endpoint with address 0x81, skipping
[ 2811.836442] usb 2-4: config 1 interface 1 altsetting 5 has a duplicate endpoint with address 0x1, skipping
[ 2811.836445] usb 2-4: config 1 interface 1 altsetting 6 has a duplicate endpoint with address 0x81, skipping
[ 2811.836447] usb 2-4: config 1 interface 1 altsetting 6 has a duplicate endpoint with address 0x1, skipping
[ 2811.836454] usb 2-4: New USB device found, idVendor=1d5c, idProduct=2000, bcdDevice= 2.00
[ 2811.836457] usb 2-4: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[ 2811.838028] usb-storage 2-4:1.3: USB Mass Storage device detected
[ 2811.838322] scsi host7: usb-storage 2-4:1.3
[ 2811.996150] usb 1-4.1: new full-speed USB device number 4 using xhci_hcd
[ 2812.117800] usb 1-4.1: New USB device found, idVendor=2000, idProduct=dd00, bcdDevice=11.16
[ 2812.117803] usb 1-4.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2812.117806] usb 1-4.1: Product: HZD100
[ 2812.117808] usb 1-4.1: Manufacturer: C&T
[ 2812.117810] usb 1-4.1: SerialNumber: U2dc0z7ws53a79rn
[ 2812.121574] input: C&T HZD100 as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.1/1-4.1:1.3/0003:2000:DD00.0003/input/input6
[ 2812.180428] hid-generic 0003:2000:DD00.0003: input,hidraw2: USB HID v1.00 Device [C&T HZD100] on usb-0000:00:14.0-4.1/input3
[ 2812.848567] scsi 7:0:0:0: Direct-Access                               A001 PQ: 0 ANSI: 4
[ 2812.849059] sd 7:0:0:0: Attached scsi generic sg3 type 0
[ 2812.849290] sd 7:0:0:0: [sdc] 2047 4096-byte logical blocks: (8.38 MB/8.00 MiB)
[ 2812.849416] sd 7:0:0:0: [sdc] Write Protect is off
[ 2812.849419] sd 7:0:0:0: [sdc] Mode Sense: 03 00 00 00
[ 2812.849541] sd 7:0:0:0: [sdc] No Caching mode page found
[ 2812.849546] sd 7:0:0:0: [sdc] Assuming drive cache: write through
[ 2812.875238]  sdc:
[ 2812.876205] sd 7:0:0:0: [sdc] Attached SCSI disk
[ 2843.836440] usb 2-4: reset SuperSpeed Gen 1 USB device number 3 using xhci_hcd
```

■ FL2000 Driver Installation
```shell
$ sudo insmod fl2000.ko
$ sudo dmesg -c
[ 3021.341650] usbcore: registered new interface driver fl2000_us
```

■ it66121.ko Installation
```shell
$ sudo insmod it66121.ko
$ sudo dmesg -c
[ 3072.069996] i2c i2c-0: Binding FL2000 master
[ 3072.071329] i2c 0-004c: Binding IT66121 component
[ 3072.583037] i2c i2c-0: Bridge attached
[ 3072.583049] i2c i2c-0: bound 0-004c (ops it66121_component_ops [it66121])
[ 3072.583057] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[ 3072.583306] [drm] Initialized fl2000_drm 0.0.1 20181001 for i2c-0 on minor 0
[ 3072.610981] i2c i2c-0: [drm] Cannot find any crtc or sizes
```

■ HDMI Connection
```shell
$ sudo dmesg -c
[sudo] ubuntu のパスワード:
[ 3132.281260] checking generic (d1000000 300000) vs hw (0 0)
[ 3132.281387] i2c i2c-

klogg, 2026-10-27:
Closing this historical installation/blank-screen report as obsolete. The driver and installation procedure have changed substantially since the Linux 5.x versions discussed here: current master targets Linux 7.0 and uses a udev-activated service to discover and instantiate the adapter's external bridge.

Please use the current build and bridge setup instructions: https://github.com/klogg/fl2000_drm#build

If the same failure persists with current master on Linux 7.0, please reopen this issue with the exact revision, adapter/output type, USB topology, full kernel log, enumerator service status/log, connector status and EDID where available, and the steps used to reproduce it. Please open a separate issue for a different failure. Native VGA is tracked in #79.

The available HDMI hardware passes visible-output and hotplug tests, but the adapters and desktop configurations in this thread have not all been retested. Closure marks the old report as obsolete rather than claiming every reported failure is fixed.

closed 2026-10-27 by klogg