klogg/fl2000_drm #44: Compilation fails at fl2000_gem_prime_vunmap/fl2000_gem_prime_vmap: Missing dma_buf_map (kernel 5.11.0)

MarcelWaldvogel, 2021-06-11:
Compiling on Ubuntu 21.04 (Kernel 5.11.0-18), I get the following errors:

```gcc
  CC [M]  /tmp/fl2000_drm/fl2000_gem.o
/tmp/fl2000_drm/fl2000_gem.c:270:10: error: initialization of ‘int (*)(struct drm_gem_object *, struct dma_buf_map *)’ from incompatible pointer type ‘void * (*)(struct drm_gem_object *)’ [-Werror=incompatible-pointer-types]
  270 |  .vmap = fl2000_gem_prime_vmap,
      |          ^~~~~~~~~~~~~~~~~~~~~
/tmp/fl2000_drm/fl2000_gem.c:270:10: note: (near initialization for ‘fl2000_gem_default_funcs.vmap’)
/tmp/fl2000_drm/fl2000_gem.c:271:12: error: initialization of ‘void (*)(struct drm_gem_object *, struct dma_buf_map *)’ from incompatible pointer type ‘void (*)(struct drm_gem_object *, void *)’ [-Werror=incompatible-pointer-types]
  271 |  .vunmap = fl2000_gem_prime_vunmap,
      |            ^~~~~~~~~~~~~~~~~~~~~~~
/tmp/fl2000_drm/fl2000_gem.c:271:12: note: (near initialization for ‘fl2000_gem_default_funcs.vunmap’)
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:287: /tmp/fl2000_drm/fl2000_gem.o] Error 1

```

It seems that an additional `dma_buf_map` parameter is required.

pedrobritobr, 2021-08-05:
Same error here, mint 20.2 kernel 5.11

tootis, 2021-09-05:
@MarcelWaldvogel Any updates on this issue?

davidnilo, 2021-09-12:
Same issue with:
- Ubuntu 20.04.3 LTS
- 5.11.0-34-generic

mmarjano, 2021-11-17:
Same issue with:

- Zorin OS 16
- 5.11.0-40-generic

derco0n, 2021-12-01:
Seems like there had been a change in the linux-kernel from version 5.10.x to 5.11.x:

5.10.x:
void *(*vmap)(struct drm_gem_object *obj);
void (*vunmap)(struct drm_gem_object *obj, void *vaddr);

5.11.x:
int (*vmap)(struct drm_gem_object *obj, struct dma_buf_map *map);
void (*vunmap)(struct drm_gem_object *obj, struct dma_buf_map *map);

it looks like from 5.11 on there is a pointer to an dma_buf_map-struct instead of a pointer to an address (assumingly integer) required...

https://elixir.bootlin.com/linux/v5.10.82/source/include/drm/drm_gem.h
https://elixir.bootlin.com/linux/v5.11.14/source/include/drm/drm_gem.h

dim6003, 2022-01-27:
Tried to fix these errors in files fl2000_gem.c and fl2000.h, but then more errors arise in fl2000_drm.c.
Not an expert at this, and I don't know this code, so stopping there, but interested in solution either. 
````
ubuntu@ubuntu:~/Documents/klogg/fl2000_drm$ make
make CHECK="/usr/bin/sparse" -C /lib/modules/5.11.0-27-generic/build M=/home/ubuntu/Documents/klogg/fl2000_drm modules
make[1] : on entre dans le répertoire « /usr/src/linux-headers-5.11.0-27-generic »
  CC [M]  /home/ubuntu/Documents/klogg/fl2000_drm/fl2000_drv.o
  CC [M]  /home/ubuntu/Documents/klogg/fl2000_drm/fl2000_registers.o
  CC [M]  /home/ubuntu/Documents/klogg/fl2000_drm/fl2000_interrupt.o
  CC [M]  /home/ubuntu/Documents/klogg/fl2000_drm/fl2000_streaming.o
  CC [M]  /home/ubuntu/Documents/klogg/fl2000_drm/fl2000_i2c.o
  CC [M]  /home/ubuntu/Documents/klogg/fl2000_drm/fl2000_gem.o
/home/ubuntu/Documents/klogg/fl2000_drm/fl2000_gem.c: In function ‘fl2000_gem_prime_vmap’:
/home/ubuntu/Documents/klogg/fl2000_drm/fl2000_gem.c:259:12: warning: returning ‘void *’ from a function with return type ‘int’ makes integer from pointer without a cast [-Wint-conversion]
  259 |  return obj->vaddr;
      |         ~~~^~~~~~~
  CC [M]  /home/ubuntu/Documents/klogg/fl2000_drm/fl2000_drm.o
/home/ubuntu/Documents/klogg/fl2000_drm/fl2000_drm.c:129:3: error: ‘struct drm_driver’ has no member named ‘gem_free_object_unlocked’
  129 |  .gem_free_object_unlocked = fl2000_gem_free,
      |   ^~~~~~~~~~~~~~~~~~~~~~~~
/home/ubuntu/Documents/klogg/fl2000_drm/fl2000_drm.c:129:30: error: initialization of ‘int (*)(struct drm_device *, struct drm_file *, uint32_t,  uint32_t,  int *)’ {aka ‘int (*)(struct drm_device *, struct drm_file *, unsigned int,  unsigned int,  int *)’} from incompatible pointer type ‘void (*)(struct drm_gem_object *)’ [-Werror=incompatible-pointer-types]
  129 |  .gem_free_object_unlocked = fl2000_gem_free,
      |                              ^~~~~~~~~~~~~~~
/home/ubuntu/Documents/klogg/fl2000_drm/fl2000_drm.c:129:30: note: (near initialization for ‘fl2000_drm_driver.prime_handle_to_fd’)
/home/ubuntu/Documents/klogg/fl2000_drm/fl2000_drm.c:130:3: error: ‘struct drm_driver’ has no member named ‘gem_vm_ops’
  130 |  .gem_vm_ops = &fl2000_gem_vm_ops,
      |   ^~~~~~~~~~
/home/ubuntu/Documents/klogg/fl2000_drm/fl2000_drm.c:130:16: error: initialization of ‘int (*)(struct drm_device *, struct drm_file *, int,  uint32_t *)’ {aka ‘int (*)(struct drm_device *, struct drm_file *, int,  unsigned int *)’} from incompatible pointer type ‘const struct vm_operations_struct *’ [-Werror=incompatible-pointer-types]
  130 |  .gem_vm_ops = &fl2000_gem_vm_ops,
      |                ^
/home/ubuntu/Documents/klogg/fl2000_drm/fl2000_drm.c:130:16: note: (near initialization for ‘fl2000_drm_driver.prime_fd_to_handle’)
/home/ubuntu/Documents/klogg/fl2000_drm/fl2000_drm.c:131:3: error: ‘struct drm_driver’ has no member named ‘gem_prime_get_sg_table’; did you mean ‘gem_prime_import_sg_table’?
  131 |  .gem_prime_get_sg_table = fl2000_gem_prime_get_sg_table,
      |   ^~~~~~~~~~~~~~~~~~~~~~
      |   gem_prime_import_sg_table
/home/ubuntu/Documents/klogg/fl2000_drm/fl2000_drm.c:131:28: error: initialization of ‘struct drm_gem_object * (*)(struct drm_device *, struct dma_buf *)’ from incompatible pointer type ‘struct sg_table * (*)(struct drm_gem_object *)’ [-Werror=incompatible-pointer-types]
  131 |  .gem_prime_get_sg_table = fl2000_gem_prime_get_sg_table,
      |                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/ubuntu/Documents/klogg/fl2000_drm/fl2000_drm.c:131:28: note: (near initialization for ‘fl2000_drm_driver.gem_prime_import’)
/home/ubuntu/Documents/klogg/fl2000_drm/fl2000_drm.c:132:3: error: ‘struct drm_driver’ has no member named ‘gem_prime_vmap’; did you mean ‘gem_prime_mmap’?
  132 |  .gem_prime_vmap = fl2000_gem_prime_vmap,
      |   ^~~~~~~~~~~~~~
      |   gem_prime_mmap
/home/ubuntu/Documents/klogg/fl2000_drm/fl2000_drm.c:132:20: error: initialization of ‘struct drm_gem_object * (*)(struct drm_device *, struct dma_buf_attachment *, struct sg_table *)’ from incompatible pointer type ‘int (*)(struct drm_gem_object *, struct dma_buf_map *)’ [-Werror=incompatible-pointer-types]
  132 |  .gem_prime_vmap = fl2000_gem_prime_vmap,
      |                    ^~~~~~~~~~~~~~~~~~~~~
/home/ubuntu/Documents/klogg/fl2000_drm/fl2000_drm.c:132:20: note: (near initialization for ‘fl2000_drm_driver.gem_prime_import_sg_table’)
/home/ubuntu/Documents/klogg/fl2000_drm/fl2000_drm.c:133:3: error: ‘struct drm_driver’ has no member named ‘gem_prime_vunmap’; did you mean ‘gem_prime_mmap’?
  133 |  .gem_prime_vunmap = fl2000_gem_prime_vunmap,
      |   ^~~~~~~~~~~~~~~~
      |   gem_prime_mmap
/home/ubuntu/Documents/klogg/fl2000_drm/fl2000_drm.c:133:22: error: initialization of ‘int (*)(struct drm_gem_object *, struct vm_area_struct *)’ from incompatible pointer type ‘void (*)(struct drm_gem_object *, struct dma_buf_map *)’ [-Werror=incompatible-pointer-types]
  133 |  .gem_prime_vunmap = fl2000_gem_prime_vunmap,
      |                      ^~~~~~~~~~~~~~~~~~~~~~~
/home/ubuntu/Documents/klogg/fl2000_drm/fl2000_drm.c:133:22: note: (near initialization for ‘fl2000_drm_driver.gem_prime_mmap’)
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:287 : /home/ubuntu/Documents/klogg/fl2000_drm/fl2000_drm.o] Erreur 1
make[1]: *** [Makefile:1848 : /home/ubuntu/Documents/klogg/fl2000_drm] Erreur 2
make[1] : on quitte le répertoire « /usr/src/linux-headers-5.11.0-27-generic »
make: *** [Makefile:21 : modules] Erreur 2
```

klogg, 2026-10-27:
The custom fl2000_gem.c implementation containing fl2000_gem_prime_vmap and fl2000_gem_prime_vunmap has been removed. Current master uses the kernel DRM shmem GEM helpers, so the incompatible callback signatures reported here no longer exist in the driver.

The current support target is Linux 7.0, with module build/load and framebuffer operation covered by the recorded tests: https://github.com/klogg/fl2000_drm/blob/c5fce004cd39b57d536b96f7d72e245ba98894b7/tests/qemu_validation_results.json

Closing this historical custom-GEM compilation report. This does not establish compatibility with Linux 5.11; please use the documented target and open a new issue if the current code fails there.

closed 2026-10-27 by klogg