klogg/fl2000_drm #46: esxi support

Docop2, 2021-06-19:
Hi Quite great project. But also, it can be quite great if you can add a support for esxi. like esxi 6.0  you passthrougt a usb port and then get the device via lsusb.. but do actually get the driver to load and bypass the vmware svga driver of the console.
Thanks  hope this can be a great feature.

klogg, 2021-06-25:
Hmm I'll try but I might need to get some new HW :) My dev pc is quite old. probably in couple months will get back to this

klogg, 2026-10-27:
This repository provides a Linux DRM/KMS driver, not a native ESXi host driver. A Linux guest can use the driver when its hypervisor successfully passes through the USB adapter, but ESXi-specific passthrough and console/display configuration have not been validated here. Current virtualization testing uses QEMU/KVM with a physical FL2000 HDMI adapter.

The current driver target and setup are documented here: https://github.com/klogg/fl2000_drm#build

Closing this historical platform-support request as outside the current implementation scope. If you encounter a driver failure inside a Linux 7.0 guest with current master, please reopen with the exact revision, guest kernel, hypervisor/passthrough configuration and full guest kernel log. A separate report should distinguish a Linux driver failure from a device that has not been successfully passed through.

closed 2026-10-27 by klogg