klogg/fl2000_drm #47: compiling on 5.12

joni909, 2021-06-30:
compiling on 5.12 resulted in this in the terminal (it didn't pass the make phase)

 ~/D/fl2000_drm-master> make
make CHECK="/usr/bin/sparse" -C /lib/modules/5.12.13-1-MANJARO/build M=/home/joni909/Downloads/fl2000_drm-master modules

make[1]: Entering directory '/usr/lib/modules/5.12.13-1-MANJARO/build'
  CC [M]  /home/joni909/Downloads/fl2000_drm-master/fl2000_gem.o
/home/joni909/Downloads/fl2000_drm-master/fl2000_gem.c: In function ‘fl2000_gem_prime_import_sg_table’:
/home/joni909/Downloads/fl2000_drm-master/fl2000_gem.c:234:15: error: implicit declaration of function ‘drm_prime_sg_to_page_addr_arrays’; did you mean ‘drm_prime_sg_to_dma_addr_array’? [-Werror=implicit-function-declaration]

  234 |         ret = drm_prime_sg_to_page_addr_arrays(sgt, obj->pages, NULL, obj->num_pages);
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |               drm_prime_sg_to_dma_addr_array
/home/joni909/Downloads/fl2000_drm-master/fl2000_gem.c: At top level:
/home/joni909/Downloads/fl2000_drm-master/fl2000_gem.c:270:17: error: initialization of ‘int (*)(struct drm_gem_object *, struct dma_buf_map *)’ from incompatible pointer type ‘void * (*)(struct drm_gem_object *)’ [-Werror=incompatible-pointer-
types]

  270 |         .vmap = fl2000_gem_prime_vmap,
      |                 ^~~~~~~~~~~~~~~~~~~~~
/home/joni909/Downloads/fl2000_drm-master/fl2000_gem.c:270:17: note: (near initialization for ‘fl2000_gem_default_funcs.vmap’)
/home/joni909/Downloads/fl2000_drm-master/fl2000_gem.c:271:19: error: initialization of ‘void (*)(struct drm_gem_object *, struct dma_buf_map *)’ from incompatible pointer type ‘void (*)(struct drm_gem_object *, void *)’ [-Werror=incompatible-pointer-types]

  271 |         .vunmap = fl2000_gem_prime_vunmap,
      |                   ^~~~~~~~~~~~~~~~~~~~~~~
/home/joni909/Downloads/fl2000_drm-master/fl2000_gem.c:271:19: note: (near initialization for ‘fl2000_gem_default_funcs.vunmap’)
cc1: some warnings being treated as errors

make[2]: *** [scripts/Makefile.build:271: /home/joni909/Downloads/fl2000_drm-master/fl2000_gem.o] Error 1
make[1]: *** [Makefile:1859: /home/joni909/Downloads/fl2000_drm-master] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.12.13-1-MANJARO/build'
make: *** [Makefile:21: modules] Error 2

felipejoribeiro, 2021-09-15:
same here, with 5.14.2-arch1-2

Skippythekangoo, 2021-11-30:
A same issue...

$ LC_ALL=C make
make CHECK="/usr/bin/sparse" -C /lib/modules/5.14.16-arch1-1/build M=/home/skippy/Documents/Yay_AUR/fl2000_drm modules
make[1]: Entering directory '/usr/lib/modules/5.14.16-arch1-1/build'
  CC [M]  /home/skippy/Documents/Yay_AUR/fl2000_drm/fl2000_gem.o
/home/skippy/Documents/Yay_AUR/fl2000_drm/fl2000_gem.c: Dans la fonction « fl2000_gem_prime_import_sg_table »:
/home/skippy/Documents/Yay_AUR/fl2000_drm/fl2000_gem.c:234:15: erreur: déclaration implicite de la fonction « drm_prime_sg_to_page_addr_arrays »; vouliez-vous utiliser « drm_prime_sg_to_dma_addr_array » ? [-Werror=implicit-function-declaration]
  234 |         ret = drm_prime_sg_to_page_addr_arrays(sgt, obj->pages, NULL, obj->num_pages);
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |               drm_prime_sg_to_dma_addr_array
/home/skippy/Documents/Yay_AUR/fl2000_drm/fl2000_gem.c: Au plus haut niveau:
/home/skippy/Documents/Yay_AUR/fl2000_drm/fl2000_gem.c:270:17: erreur: initialisation de « int (*)(struct drm_gem_object *, struct dma_buf_map *) » depuis le type pointeur « void * (*)(struct drm_gem_object *) » qui est incompatible [-Werror=incompatible-pointer-types]
  270 |         .vmap = fl2000_gem_prime_vmap,
      |                 ^~~~~~~~~~~~~~~~~~~~~
/home/skippy/Documents/Yay_AUR/fl2000_drm/fl2000_gem.c:270:17: note: (près de l'initialisation de « fl2000_gem_default_funcs.vmap »)
/home/skippy/Documents/Yay_AUR/fl2000_drm/fl2000_gem.c:271:19: erreur: initialisation de « void (*)(struct drm_gem_object *, struct dma_buf_map *) » depuis le type pointeur « void (*)(struct drm_gem_object *, void *) » qui est incompatible [-Werror=incompatible-pointer-types]
  271 |         .vunmap = fl2000_gem_prime_vunmap,
      |                   ^~~~~~~~~~~~~~~~~~~~~~~
/home/skippy/Documents/Yay_AUR/fl2000_drm/fl2000_gem.c:271:19: note: (près de l'initialisation de « fl2000_gem_default_funcs.vunmap »)
cc1 : certains avertissements sont traités comme des erreurs
make[2]: *** [scripts/Makefile.build:271 : /home/skippy/Documents/Yay_AUR/fl2000_drm/fl2000_gem.o] Erreur 1
make[1]: *** [Makefile:1858: /home/skippy/Documents/Yay_AUR/fl2000_drm] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.14.16-arch1-1/build'
make: *** [Makefile:21: modules] Error 2

MBaltz, 2021-12-01:
__Same here :,(__
_Please, fix that!_

`kernel: 5.15.5-arch1-1`

[image]

<details><summary>See this error in text:</summary>

```bash
make CHECK="/usr/bin/sparse" -C /lib/modules/5.15.5-arch1-1/build M=/tmp/b/fl2000_drm modules
make[1]: Entrando no diretório '/usr/lib/modules/5.15.5-arch1-1/build'

  CC [M]  /tmp/b/fl2000_drm/fl2000_drv.o
  CC [M]  /tmp/b/fl2000_drm/fl2000_registers.o
  CC [M]  /tmp/b/fl2000_drm/fl2000_interrupt.o
  CC [M]  /tmp/b/fl2000_drm/fl2000_streaming.o
  CC [M]  /tmp/b/fl2000_drm/fl2000_i2c.o
  CC [M]  /tmp/b/fl2000_drm/fl2000_gem.o
/tmp/b/fl2000_drm/fl2000_gem.c: In function ‘fl2000_gem_prime_import_sg_table’:
/tmp/b/fl2000_drm/fl2000_gem.c:234:15: error: implicit declaration of function ‘drm_prime_sg_to_page_addr_arrays’; did you mean ‘drm_prime_sg_to_dma_addr_array’? [-Werror=implicit-function-declaration]
  234 |         ret = drm_prime_sg_to_page_addr_arrays(sgt, obj->pages, NULL, obj->num_pages);
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |               drm_prime_sg_to_dma_addr_array
/tmp/b/fl2000_drm/fl2000_gem.c: At top level:
/tmp/b/fl2000_drm/fl2000_gem.c:270:17: error: initialization of ‘int (*)(struct drm_gem_object *, struct dma_buf_map *)’ from incompatible pointer type ‘void * (*)(struct drm_gem_object *)’ [-Werror=incompatible-pointer-types]
  270 |         .vmap = fl2000_gem_prime_vmap,
      |                 ^~~~~~~~~~~~~~~~~~~~~
/tmp/b/fl2000_drm/fl2000_gem.c:270:17: note: (near initialization for ‘fl2000_gem_default_funcs.vmap’)
/tmp/b/fl2000_drm/fl2000_gem.c:271:19: error: initialization of ‘void (*)(struct drm_gem_object *, struct dma_buf_map *)’ from incompatible pointer type ‘void (*)(struct drm_gem_object *, void *)’ [-Werror=incompatible-pointer-types]
  271 |         .vunmap = fl2000_gem_prime_vunmap,
      |                   ^~~~~~~~~~~~~~~~~~~~~~~
/tmp/b/fl2000_drm/fl2000_gem.c:271:19: note: (near initialization for ‘fl2000_gem_default_funcs.vunmap’)
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:277: /tmp/b/fl2000_drm/fl2000_gem.o] Erro 1
make[1]: *** [Makefile:1868: /tmp/b/fl2000_drm] Erro 2
make[1]: Saindo do diretório '/usr/lib/modules/5.15.5-arch1-1/build'

make: *** [Makefile:21: modules] Erro 2

```

</details>

Vini98br, 2022-07-14:
Same here

kernel: 5.15.11-76051511-generic

Vini98br, 2022-07-18:
@mati75 this commit fixed the `make` command errors but when running the next command of the tutorial: `insmod fl2000.ko && insmod it66121.ko`, it gave me the following error:

[image]

And when running only `sudo insmod it66121.ko` it crashed my entire computer

shanez1215, 2022-08-06:
> 

This happened to me too

petereon, 2023-07-29:
> Fix: [rhgndf@7be98c3](https://github.com/rhgndf/fl2000_drm/commit/7be98c3bc37286a768ef7fd72b2df328cf9779a6)

This fix is incomplete and I don't recommend trying to use this as of yet, I am currently in process of furthering this development as a side project.

klogg, 2026-10-27:
The reported failures involve drm_prime_sg_to_page_addr_arrays and the vmap/vunmap callbacks in the former custom fl2000_gem.c implementation. That implementation has been removed; current master uses the kernel DRM shmem GEM helpers instead.

Current validation covers module build/load, framebuffer operation, USB reset, unbind/rebind and removal with open DRM files on the Linux 7.0 target: https://github.com/klogg/fl2000_drm/blob/c5fce004cd39b57d536b96f7d72e245ba98894b7/tests/qemu_validation_results.json

The compile-only patches discussed above are not needed for current master. Linux 5.12 and the other historical kernels in this thread are outside the current support target. Please report any current Linux 7.0 failure separately with the exact revision, reproduction steps and full build or kernel log.

closed 2026-10-27 by klogg