klogg/fl2000_drm #48: not working on Gentoo with kernel 5.10.27

mgcyung, 2021-07-01:
I am trying the driver on Gentoo with kernel 5.10.27. The building was
successful. `quirks=1D5C:2000:USB_QUIRK_NO_LPM` is added to boot
params. However, it didn't work. There were errors in the dmesg
`
[   30.420909] usb 3-2: new high-speed USB device number 3 using xhci_hcd
[   30.549232] usb 3-2: New USB device found, idVendor=14cd, idProduct=8601, bcdDevice= 0.00
[   30.549237] usb 3-2: New USB device strings: Mfr=1, Product=3, SerialNumber=0
[   30.549240] usb 3-2: Product: USB 2.0 Hub
[   30.549242] usb 3-2: Manufacturer: USB Device
[   30.600764] hub 3-2:1.0: USB hub found
[   30.601042] hub 3-2:1.0: 4 ports detected
[   30.661055] usb 4-2: new SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[   30.674145] usb 4-2: config 1 interface 1 altsetting 0 has a duplicate endpoint with address 0x81, skipping
[   30.674150] usb 4-2: config 1 interface 1 altsetting 0 has a duplicate endpoint with address 0x1, skipping
[   30.674152] usb 4-2: config 1 interface 1 altsetting 1 has a duplicate endpoint with address 0x81, skipping
[   30.674154] usb 4-2: config 1 interface 1 altsetting 1 has a duplicate endpoint with address 0x1, skipping
[   30.674157] usb 4-2: config 1 interface 1 altsetting 2 has a duplicate endpoint with address 0x81, skipping
[   30.674159] usb 4-2: config 1 interface 1 altsetting 2 has a duplicate endpoint with address 0x1, skipping
[   30.674161] usb 4-2: config 1 interface 1 altsetting 3 has a duplicate endpoint with address 0x81, skipping
[   30.674163] usb 4-2: config 1 interface 1 altsetting 3 has a duplicate endpoint with address 0x1, skipping
[   30.674165] usb 4-2: config 1 interface 1 altsetting 4 has a duplicate endpoint with address 0x81, skipping
[   30.674167] usb 4-2: config 1 interface 1 altsetting 4 has a duplicate endpoint with address 0x1, skipping
[   30.674169] usb 4-2: config 1 interface 1 altsetting 5 has a duplicate endpoint with address 0x81, skipping
[   30.674171] usb 4-2: config 1 interface 1 altsetting 5 has a duplicate endpoint with address 0x1, skipping
[   30.674173] usb 4-2: config 1 interface 1 altsetting 6 has a duplicate endpoint with address 0x81, skipping
[   30.674175] usb 4-2: config 1 interface 1 altsetting 6 has a duplicate endpoint with address 0x1, skipping
[   30.674181] usb 4-2: New USB device found, idVendor=1d5c, idProduct=2000, bcdDevice= 2.00
[   30.674183] usb 4-2: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[   30.680952] usb-storage 4-2:1.3: USB Mass Storage device detected
[   30.681178] scsi host2: usb-storage 4-2:1.3
[   30.887874] usb 3-2.1: new full-speed USB device number 4 using xhci_hcd
[   30.996568] usb 3-2.1: New USB device found, idVendor=2000, idProduct=dd00, bcdDevice=11.16
[   30.996572] usb 3-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   30.996573] usb 3-2.1: Product: HZD100
[   30.996575] usb 3-2.1: Manufacturer: C&T
[   30.996576] usb 3-2.1: SerialNumber: U2dc0z7ws53a79rn
[   31.057962] input: C&T HZD100 as /devices/pci0000:00/0000:00:08.1/0000:04:00.4/usb3/3-2/3-2.1/3-2.1:1.3/0003:2000:DD00.0003/input/input28
[   31.110216] hid-generic 0003:2000:DD00.0003: input,hidraw2: USB HID v1.00 Device [C&T HZD100] on usb-0000:04:00.4-2.1/input3
[   31.236662] mc: Linux media interface: v0.10
[   31.437959] usbcore: registered new interface driver snd-usb-audio
[   31.512210] usbcore: registered new interface driver uas
[   31.708198] scsi 2:0:0:0: Direct-Access                               A001 PQ: 0 ANSI: 4
[   31.708651] sd 2:0:0:0: Attached scsi generic sg0 type 0
[   31.708882] sd 2:0:0:0: [sda] 2047 4096-byte logical blocks: (8.38 MB/8.00 MiB)
[   31.709035] sd 2:0:0:0: [sda] Write Protect is off
[   31.709037] sd 2:0:0:0: [sda] Mode Sense: 03 00 00 00
[   31.709155] sd 2:0:0:0: [sda] No Caching mode page found
[   31.709157] sd 2:0:0:0: [sda] Assuming drive cache: write through
[   31.721323]  sda:
[   31.722453] sd 2:0:0:0: [sda] Attached SCSI disk
[   47.325102] usb 4-2: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[   62.685101] usb 4-2: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[   73.445169] usbcore: registered new interface driver fl2000_usb
[   78.044969] usb 4-2: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[   79.814204] i2c i2c-0: I2C transfer failed (-121)
[   79.814394] i2c i2c-10: Binding FL2000 master
[   79.814968] i2c 0-004c: Binding IT66121 component
[   79.922577] i2c i2c-10: Bridge attached
[   79.922600] i2c i2c-10: bound 0-004c (ops it66121_component_ops [it66121])
[   79.923308] [drm] Initialized fl2000_drm 0.0.1 20181001 for i2c-10 on minor 1
[   79.935645] i2c i2c-10: Cannot get monitor status (-121)
[   79.935793] i2c i2c-10: [drm] Cannot find any crtc or sizes
[   79.956820] i2c i2c-10: Cannot get monitor status (-121)
[   79.956927] i2c i2c-10: [drm] Cannot find any crtc or sizes
[   80.028391] i2c i2c-10: Cannot read interrupt status (-121)
[   91.970773] udevd[2434]: worker [4956] /devices/pci0000:00/0000:00:08.1/0000:04:00.4/usb4/4-2/4-2:1.2 is taking a long time
[   92.521076] udevd[4956]: slow: 'lmt-udev force' [6379]
[   93.287887] usb 4-2: Stopping interrupts
[   93.405234] usb 4-2: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[   93.421269] i2c i2c-10: Binding FL2000 master
[... 35 lines cut ...]
[   93.422731]  really_probe+0xf0/0x400
[   93.422735]  driver_probe_device+0xe1/0x150
[   93.422738]  ? driver_allows_async_probing+0x50/0x50
[   93.422741]  bus_for_each_drv+0x6a/0xa0
[   93.422744]  __device_attach+0xc7/0x140
[   93.422748]  rebind_marked_interfaces.isra.0+0x62/0x90
[   93.422753]  usb_reset_device+0x1af/0x250
[   93.422761]  usb_stor_port_reset+0x52/0x70 [usb_storage]
[   93.422767]  usb_stor_invoke_transport+0x20a/0x510 [usb_storage]
[   93.422773]  ? try_to_wake_up+0x1ad/0x4d0
[   93.422777]  ? __prepare_to_swait+0x4b/0x70
[   93.422782]  ? wait_for_completion_interruptible+0xb1/0x120
[   93.422787]  usb_stor_control_thread+0x185/0x280 [usb_storage]
[   93.422791]  ? storage_probe+0x280/0x280 [usb_storage]
[   93.422796]  kthread+0xfe/0x140
[   93.422799]  ? kthread_park+0x90/0x90
[   93.422804]  ret_from_fork+0x22/0x30
[   93.422808] ---[ end trace 8d4c50ab56873abf ]---
[   93.422823] [drm:drm_atomic_helper_shutdown [drm_kms_helper]] *ERROR* Disabling all crtc's during unload failed with -12
[   93.422891] i2c i2c-10: master bind failed: -16
[   93.422905] usb 4-2: Cannot register component master (-16)
[   93.422919] fl2000_usb: probe of 4-2:1.2 failed with error -16
[   93.460110] i2c i2c-10: Cannot get monitor status (-121)
[   93.460364] i2c i2c-10: Cannot get monitor status (-121)
`

skyEch0, 2021-07-24:
Any update on this? facing the exact issue with Ubuntu 20.10, Kernel 5.8

klogg, 2026-10-27:
Closing this historical Linux 5.10 report as obsolete following substantial changes to USB probe/teardown, bridge attachment, I2C handling and DRM lifetime management. Current master targets Linux 7.0. The component-binding and shutdown failure paths shown in the old log have been reworked, with reset, rebind and removal tests recorded for the current code.

Current setup: https://github.com/klogg/fl2000_drm#build
Recorded validation: https://github.com/klogg/fl2000_drm/blob/c5fce004cd39b57d536b96f7d72e245ba98894b7/tests/qemu_validation_results.json

If the same failure persists with current master on Linux 7.0, please reopen this issue with the exact revision, kernel configuration, adapter/output type, USB topology, reproduction steps, full kernel log and enumerator service log. Native VGA is tracked in #79.

The original Gentoo setup has not been retested; closure marks the historical report as obsolete rather than claiming that configuration has been validated.

closed 2026-10-27 by klogg