klogg/fl2000_drm #50: Doesn't work Ubuntu

killerdevildog11, 2021-09-24:
fl2000_drm$ make
make CHECK="/usr/bin/sparse" -C /lib/modules/5.11.0-36-generic/build M=/home/killer/Documents/00002_GIT/fl2000_drm modules
make[1]: Entering directory '/usr/src/linux-headers-5.11.0-36-generic'

  ERROR: Kernel configuration is invalid.
         include/generated/autoconf.h or include/config/auto.conf are missing.
         Run 'make oldconfig && make prepare' on kernel src to fix it.

make[1]: *** [Makefile:745: include/config/auto.conf] Error 1
make[1]: Leaving directory '/usr/src/linux-headers-5.11.0-36-generic'
make: *** [Makefile:21: modules] Error 2

then go to source it does this 

:/usr/src/linux-hwe-5.11-headers-5.11.0-36$ sudo make oldconfig && make prepare
[sudo] password for killer: 
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/confdata.o
  HOSTCC  scripts/kconfig/expr.o
  LEX     scripts/kconfig/lexer.lex.c
flex: -Cf/-CF and %option yylineno are incompatible
make[1]: *** [scripts/Makefile.host:9: scripts/kconfig/lexer.lex.c] Error 1
make: *** [Makefile:629: oldconfig] Error 2

ping-mee, 2021-10-13:
I have the same Problem :(

HRIDOY-BUZZ, 2021-11-22:
Experiencing the Same issue in Kernel 5.13
Anyone solved it yet?

klogg, 2026-10-27:
The reported failure occurs while checking the selected kernel build tree, before compiling driver sources. Missing generated configuration files indicate that the selected headers/build tree is incomplete or not prepared. Distribution header packages should be installed for the exact target kernel; running configuration commands inside a partial packaged header tree is not the supported installation procedure.

Current build and packaging instructions target Linux 7.0: https://github.com/klogg/fl2000_drm#build

Closing this historical Linux 5.11 environment report as outside the current support scope. If the documented Linux 7.0 setup fails, please open a new issue with the exact revision, header package versions and complete build log.

closed 2026-10-27 by klogg