klogg/fl2000_drm #51: Compile issue - "No rule to make target"

andrewcapybara, 2022-01-27:
Sorry, I'm a bit of a noob, so maybe I should know my way out of this.
It seems Mr Make can't find his way into a directory:

```
make[1]: Entering directory '/usr/src/linux-headers-5.15.15-76051515-generic'
make[2]: *** No rule to make target 'arch/x86/entry/syscalls/syscall_32.tbl', needed by 'arch/x86/include/generated/uapi/asm/unistd_32.h'. Stop.
make[1]: *** [arch/x86/Makefile:213: archheaders] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.15-76051515-generic'
make: *** [Makefile:21: modules] Error 2
```

What do I need to do to continue the compile?

juanmoura, 2022-04-22:
same error here

shanez1215, 2022-08-06:
I was able to fix this by opening makefile and replacing all instances of M=$(PWD) with M=$(shell pwd)

joelnoliveira, 2023-11-01:
> I was able to fix this by opening makefile and replacing all instances of M=$(PWD) with M=$(shell pwd)

Where is that makefile located?

abysslover, 2024-04-19:
May be /lib/modules/$(uname -r)/build

klogg, 2026-10-27:
The reported syscall_32.tbl failure occurs in the kernel header/build tree before driver compilation. Current Makefile uses M="$(CURDIR)" to identify the external module directory, replacing the older PWD-based invocation discussed in this thread.

Please use the complete headers matching the target kernel and the current build instructions: https://github.com/klogg/fl2000_drm#build

The project now targets Linux 7.0. Closing this historical Linux 5.15 environment report as outside the current support scope, without claiming that its missing kernel build files were fixed by this repository. If the documented Linux 7.0 setup fails, please provide the exact revision, header package versions and complete build log in a new issue.

closed 2026-10-27 by klogg