klogg/fl2000_drm #52: installation problem

saikat0003, 2022-02-10:
saikat@saikat-ROG-Strix:~/Downloads/fl2000_drm-master$ make
make CHECK="/usr/bin/sparse" -C /lib/modules/5.13.0-28-generic/build M=/home/saikat/Downloads/fl2000_drm-master modules
make[1]: Entering directory '/usr/src/linux-headers-5.13.0-28-generic'
  CC [M]  /home/saikat/Downloads/fl2000_drm-master/fl2000_gem.o
/home/saikat/Downloads/fl2000_drm-master/fl2000_gem.c: In function ‘fl2000_gem_prime_import_sg_table’:
/home/saikat/Downloads/fl2000_drm-master/fl2000_gem.c:234:8: error: implicit declaration of function ‘drm_prime_sg_to_page_addr_arrays’; did you mean ‘drm_prime_sg_to_dma_addr_array’? [-Werror=implicit-function-declaration]
  234 |  ret = drm_prime_sg_to_page_addr_arrays(sgt, obj->pages, NULL, obj->num_pages);
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |        drm_prime_sg_to_dma_addr_array
/home/saikat/Downloads/fl2000_drm-master/fl2000_gem.c: At top level:
/home/saikat/Downloads/fl2000_drm-master/fl2000_gem.c:270:10: error: initialization of ‘int (*)(struct drm_gem_object *, struct dma_buf_map *)’ from incompatible pointer type ‘void * (*)(struct drm_gem_object *)’ [-Werror=incompatible-pointer-types]
  270 |  .vmap = fl2000_gem_prime_vmap,
      |          ^~~~~~~~~~~~~~~~~~~~~
/home/saikat/Downloads/fl2000_drm-master/fl2000_gem.c:270:10: note: (near initialization for ‘fl2000_gem_default_funcs.vmap’)
/home/saikat/Downloads/fl2000_drm-master/fl2000_gem.c:271:12: error: initialization of ‘void (*)(struct drm_gem_object *, struct dma_buf_map *)’ from incompatible pointer type ‘void (*)(struct drm_gem_object *, void *)’ [-Werror=incompatible-pointer-types]
  271 |  .vunmap = fl2000_gem_prime_vunmap,
      |            ^~~~~~~~~~~~~~~~~~~~~~~
/home/saikat/Downloads/fl2000_drm-master/fl2000_gem.c:271:12: note: (near initialization for ‘fl2000_gem_default_funcs.vunmap’)
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:281: /home/saikat/Downloads/fl2000_drm-master/fl2000_gem.o] Error 1
make[1]: *** [Makefile:1879: /home/saikat/Downloads/fl2000_drm-master] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.13.0-28-generic'
make: *** [Makefile:21: modules] Error 2

petereon, 2022-03-20:
This is a duplicate of #47 , seems like maintainers no longer works on this. I have worked on it a bit in my fork and it now compiles but I did not get it to work yet.

klogg, 2026-10-27:
This is the same custom-GEM compilation failure discussed in #47. The fl2000_gem.c implementation containing drm_prime_sg_to_page_addr_arrays and the incompatible vmap/vunmap callbacks has been removed. Current master uses the kernel DRM shmem GEM helpers instead.

Build and loading instructions for the current Linux 7.0 target: https://github.com/klogg/fl2000_drm#build

The historical Linux 5.13 kernel reported here is outside the current support target. Please open a new issue with the exact revision and complete build log if current master fails on the documented target.

closed 2026-10-27 by klogg