klogg/fl2000_drm #53: kernel 5.16

jlhdzl, 2022-07-26:
Apply petereon patch

But when compiling it gives the following error:

  LD [M]  /usr/src/fl2000_drm-master/fl2000.o
  MODPOST /usr/src/fl2000_drm-master/Module.symvers
ERROR: modpost: module fl2000 uses symbol dma_buf_begin_cpu_access from namespace DMA_BUF, but does not import it.
ERROR: modpost: module fl2000 uses symbol dma_buf_end_cpu_access from namespace DMA_BUF, but does not import it.
make[2]: *** [scripts/Makefile.modpost:134: /usr/src/fl2000_drm-master/Module.symvers] Error 1
make[2]: *** Se borra el archivo '/usr/src/fl2000_drm-master/Module.symvers'
make[1]: *** [Makefile:1767: modules] Error 2
make[1]: se sale del directorio '/usr/src/linux-headers-5.16.15-051615-generic'
make: *** [Makefile:21: modules] Error 2

jlhdzl, 2022-07-26:
make CHECK="/usr/bin/sparse" -C /lib/modules/5.16.15-051615-generic/build M=/usr/src/fl2000_drm-master modules
make[1]: se entra en el directorio '/usr/src/linux-headers-5.16.15-051615-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (Ubuntu 9.4.0-1ubuntu1~20.04) 9.4.0
  You are using:           gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0

klogg, 2026-10-27:
The DMA_BUF namespace failure reported here occurred while building an older driver with a compile-only compatibility patch. Current master has replaced the custom GEM implementation with the kernel DRM shmem GEM helpers, and the current modules pass modpost, build and load testing on Linux 7.0.

Build instructions: https://github.com/klogg/fl2000_drm#build

Recorded validation: https://github.com/klogg/fl2000_drm/blob/c5fce004cd39b57d536b96f7d72e245ba98894b7/tests/qemu_validation_results.json

Closing this historical build report. Linux 5.16 compatibility is outside the current support target. If a namespace error occurs with current master on Linux 7.0, please open a new issue with the exact revision and complete build log.

closed 2026-10-27 by klogg