klogg/fl2000_drm #54: error in make kernel 5.15.0-48

MartinCastellano, 2022-10-01:
hi when i do make i have this proble 

`/home/mcastellano/Descargas/FL2000/fl2000_drm/fl2000_gem.c: In function ‘fl2000_gem_prime_import_sg_table’:
/home/mcastellano/Descargas/FL2000/fl2000_drm/fl2000_gem.c:234:8: error: implicit declaration of function ‘drm_prime_sg_to_page_addr_arrays’; did you mean ‘drm_prime_sg_to_dma_addr_array’? [-Werror=implicit-function-declaration]
  234 |  ret = drm_prime_sg_to_page_addr_arrays(sgt, obj->pages, NULL, obj->num_pages);
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |        drm_prime_sg_to_dma_addr_array
/home/mcastellano/Descargas/FL2000/fl2000_drm/fl2000_gem.c: At top level:
/home/mcastellano/Descargas/FL2000/fl2000_drm/fl2000_gem.c:270:10: error: initialization of ‘int (*)(struct drm_gem_object *, struct dma_buf_map *)’ from incompatible pointer type ‘void * (*)(struct drm_gem_object *)’ [-Werror=incompatible-pointer-types]
  270 |  .vmap = fl2000_gem_prime_vmap,
      |          ^~~~~~~~~~~~~~~~~~~~~
/home/mcastellano/Descargas/FL2000/fl2000_drm/fl2000_gem.c:270:10: note: (near initialization for ‘fl2000_gem_default_funcs.vmap’)
/home/mcastellano/Descargas/FL2000/fl2000_drm/fl2000_gem.c:271:12: error: initialization of ‘void (*)(struct drm_gem_object *, struct dma_buf_map *)’ from incompatible pointer type ‘void (*)(struct drm_gem_object *, void *)’ [-Werror=incompatible-pointer-types]
  271 |  .vunmap = fl2000_gem_prime_vunmap,
      |            ^~~~~~~~~~~~~~~~~~~~~~~
/home/mcastellano/Descargas/FL2000/fl2000_drm/fl2000_gem.c:271:12: note: (near initialization for ‘fl2000_gem_default_funcs.vunmap’)
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:297: /home/mcastellano/Descargas/FL2000/fl2000_drm/fl2000_gem.o] Error 1
make[1]: *** [Makefile:1884: /home/mcastellano/Descargas/FL2000/fl2000_drm] Error 2
make[1]: se sale del directorio '/usr/src/linux-headers-5.15.0-48-generic'
make: *** [Makefile:21: modules] Error 2`

so its seems that the function drm_prime_sg_to_page_addr_arrays its not defined ..and others

temporary999, 2022-10-19:
Hello
I have fixed the  drm_prime_sg_to_page_addr_arrays issue by "https://github.com/Xilinx/XRT/issues/5943%22's hint.

yanlobianchi, 2022-10-27:
@temporary999 please share your code after fixed

anvalenciao, 2022-10-31:
### Error

> fl2000_drm/fl2000_gem.c:234:8: error: implicit declaration of function ‘drm_prime_sg_to_page_addr_arrays’; did you mean ‘drm_prime_sg_to_dma_addr_array’? [-Werror=implicit-function-declaration] [...]

### Solution
**Line 234 of the file fl2000_gem.c:**
`$ nano +234 fl2000_gem.c`
**Change:**
`ret = drm_prime_sg_to_page_addr_arrays(sgt, obj->pages, NULL, obj->num_pages);`
**To:**
`ret = drm_prime_sg_to_page_array(sgt, obj->pages, obj->num_pages);`

klogg, 2026-10-27:
The reported drm_prime_sg_to_page_addr_arrays and vmap/vunmap failures come from the former custom fl2000_gem.c implementation. That implementation has been removed; current master uses the kernel DRM shmem GEM helpers. The local substitutions discussed above are no longer needed.

Current build and loading instructions: https://github.com/klogg/fl2000_drm#build

The supported target is Linux 7.0, where module build/load and framebuffer operation have been tested. This does not establish Linux 5.15 compatibility. Please open a new issue with the exact revision and full build log if current master fails on the documented target. See also #47.

closed 2026-10-27 by klogg