klogg/fl2000_drm #55: compile issues

temporary999, 2022-10-19:
Hello
I have fixed the  drm_prime_sg_to_page_addr_arrays issue by "https://github.com/Xilinx/XRT/issues/5943"'s hint.

Now I have this pointer issue
                                                                                         
make CHECK="/usr/bin/sparse" -C /lib/modules/6.0.2-zen1-1-zen/build M=/home/user/dev/git/fl2000_drm modules
make[1] : on entre dans le répertoire « /usr/lib/modules/6.0.2-zen1-1-zen/build »
  CC [M]  /home/user/dev/git/fl2000_drm/fl2000_gem.o
/home/user/dev/git/fl2000_drm/fl2000_gem.c:273:17: erreur: initialisation de « int (*)(struct drm_gem_object *, struct iosys_map *) » depuis le type pointeur « void * (*)(struct drm_gem_object *) » qui est incompatible [-Werror=incompatible-pointer-types]
  273 |         .vmap = fl2000_gem_prime_vmap,
      |                 ^~~~~~~~~~~~~~~~~~~~~
/home/user/dev/git/fl2000_drm/fl2000_gem.c:273:17: note: (près de l'initialisation de « fl2000_gem_default_funcs.vmap »)
/home/user/dev/git/fl2000_drm/fl2000_gem.c:274:19: erreur: initialisation de « void (*)(struct drm_gem_object *, struct iosys_map *) » depuis le type pointeur « void (*)(struct drm_gem_object *, void *) » qui est incompatible [-Werror=incompatible-pointer-types]
  274 |         .vunmap = fl2000_gem_prime_vunmap,
      |                   ^~~~~~~~~~~~~~~~~~~~~~~
/home/user/dev/git/fl2000_drm/fl2000_gem.c:274:19: note: (près de l'initialisation de « fl2000_gem_default_funcs.vunmap »)
cc1 : certains avertissements sont traités comme des erreurs
make[2]: *** [scripts/Makefile.build:249 : /home/user/dev/git/fl2000_drm/fl2000_gem.o] Erreur 1
make[1]: *** [Makefile:1849 : /home/user/dev/git/fl2000_drm] Erreur 2
make[1] : on quitte le répertoire « /usr/lib/modules/6.0.2-zen1-1-zen/build »
make: *** [Makefile:23 : modules] Erreur 2

Thank you again :)

temporary999, 2022-10-19:
I have fixed the pointer .. kinda,
but it seams now that the DRM interface is not compatible with kernel post 5.11 or 5.18 .

best regards

bigrck64, 2023-10-25:
How did you fix them ? :)
I'm also trying to make it work (under kernel 5.15)

remi75, 2023-10-25:
I hunted down the C language errors, that is standard&classic, as said it didn't work anyway ...
we can do nothing without a real Linux driver developer 
And
follow the kernel's architecture to update it correctly ...

bigrck64, 2023-10-26:
Hi Remi,

This fork seems to add VGA support and newer kernel support :)
https://github.com/rhgndf/fl2000_drm

I can compile it, but no way to "insmod" it into the kernel.

remi75, 2023-10-26:
have you filed a bug report ?
last work 2 weeks ago, that is very promising ! thank you for spotting it

bigrck64, 2023-10-26:
Yes I created one issue there :)

klogg, 2026-10-27:
The incompatible iosys_map vmap/vunmap callbacks in this report belonged to the custom fl2000_gem.c implementation. That implementation has been removed. Current master uses the kernel DRM shmem GEM helpers, so the reported callback mismatch and the preceding drm_prime_sg_to_page_addr_arrays error no longer apply.

Current build and loading instructions: https://github.com/klogg/fl2000_drm#build

The supported and tested target is Linux 7.0; this does not establish compatibility with Linux 6.0 or the other historical kernels discussed here. Native VGA remains unimplemented and is tracked separately in #79. Please open a new issue if current master fails on the documented target, including the exact revision and complete logs.

closed 2026-10-27 by klogg