klogg/fl2000_drm #57: Cannot find any crtc or sizes

ftoledo, 2023-04-27:
I sussceful build on debian 11:
Linux ragnarok 5.10.0-21-amd64 #1 SMP Debian 5.10.162-1 (2023-01-21) x86_64 GNU/Linux

modules load, but get thes

[41443.901209] i2c i2c-14: Binding FL2000 master
[41443.902365] i2c 0-004c: Binding IT66121 component
[41444.024292] i2c i2c-14: Bridge attached
[41444.024301] i2c i2c-14: bound 0-004c (ops it66121_component_ops [it66121])
[41444.024583] [drm] Initialized fl2000_drm 0.0.1 20181001 for i2c-14 on minor 2
[41444.044344] i2c i2c-14: Cannot get monitor status (-6)
[41444.044877] i2c i2c-14: [drm] Cannot find any crtc or sizes
[41444.127736] i2c i2c-14: Cannot read interrupt status (-6)
[41444.558517] i2c i2c-14: Cannot get monitor status (-6)
[41444.559197] i2c i2c-14: [drm] Cannot find any crtc or sizes

thenightterorx, 2024-04-12:
are you able to get any display output?

ftoledo, 2024-04-13:
sorry i dont hae debian 11 anymore. NOw at debian 12 i cannot build
```
ragnarok@ragnarok:~/src/fl2000_drm$ make
make CHECK="/usr/bin/sparse" -C /lib/modules/6.1.0-20-amd64/build M=/home/ragnarok/src/fl2000_drm modules
make[1]: se entra en el directorio '/usr/src/linux-headers-6.1.0-20-amd64'
  CC [M]  /home/ragnarok/src/fl2000_drm/fl2000_drv.o
In file included from /home/ragnarok/src/fl2000_drm/fl2000_drv.c:7:
/home/ragnarok/src/fl2000_drm/fl2000.h:33:10: fatal error: drm/drm_fbdev_generic.h: No existe el fichero o el directorio
   33 | #include <drm/drm_fbdev_generic.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [/usr/src/linux-headers-6.1.0-20-common/scripts/Makefile.build:255: /home/ragnarok/src/fl2000_drm/fl2000_drv.o] Error 1
make[1]: *** [/usr/src/linux-headers-6.1.0-20-common/Makefile:2035: /home/ragnarok/src/fl2000_drm] Error 2
make[1]: se sale del directorio '/usr/src/linux-headers-6.1.0-20-amd64'
make: *** [Makefile:20: modules] Error 2
ragnarok@ragnarok:~/src/fl2000_drm$ uname -a
Linux ragnarok 6.1.0-20-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.85-1 (2024-04-11) x86_64 GNU/Linux

```

thenightterorx, 2024-04-13:
im using arch but I'll see if I can get it working for you using the older 6.1 lts kernel. It might just require a few patches.

thenightterorx, 2024-04-13:
it seems to not complain if you do the sketchy thing of manually downloading that file from the kernel sources and putting it in your build directory then changing the include statement to comments. I still get the same error though as I do on linux versions 6.6 and 6.8 (which is configure crtc 6 failed in xrandr, could not set tx mode in dmesg.

ftoledo, 2024-04-13:
i also try backport 6.5 kernel, and still no success
```
ragnarok@ragnarok:~/src/fl2000_drm-rhgndf$ make
make CHECK="/usr/bin/sparse" -C /lib/modules/6.5.0-0.deb12.4-amd64/build M=/home/ragnarok/src/fl2000_drm-rhgndf modules
make[1]: se entra en el directorio '/usr/src/linux-headers-6.5.0-0.deb12.4-amd64'
  CC [M]  /home/ragnarok/src/fl2000_drm-rhgndf/fl2000_drv.o
  CC [M]  /home/ragnarok/src/fl2000_drm-rhgndf/fl2000_registers.o
  CC [M]  /home/ragnarok/src/fl2000_drm-rhgndf/fl2000_interrupt.o
  CC [M]  /home/ragnarok/src/fl2000_drm-rhgndf/fl2000_streaming.o
  CC [M]  /home/ragnarok/src/fl2000_drm-rhgndf/fl2000_connector.o
  CC [M]  /home/ragnarok/src/fl2000_drm-rhgndf/fl2000_i2c.o
  CC [M]  /home/ragnarok/src/fl2000_drm-rhgndf/fl2000_drm.o
  LD [M]  /home/ragnarok/src/fl2000_drm-rhgndf/fl2000.o
  MODPOST /home/ragnarok/src/fl2000_drm-rhgndf/Module.symvers
  CC [M]  /home/ragnarok/src/fl2000_drm-rhgndf/fl2000.mod.o
  LD [M]  /home/ragnarok/src/fl2000_drm-rhgndf/fl2000.ko
  BTF [M] /home/ragnarok/src/fl2000_drm-rhgndf/fl2000.ko
/bin/sh: 1: ./tools/bpf/resolve_btfids/resolve_btfids: not found
make[3]: *** [/usr/src/linux-headers-6.5.0-0.deb12.4-common/scripts/Makefile.modfinal:63: /home/ragnarok/src/fl2000_drm-rhgndf/fl2000.ko] Error 127
make[3]: *** Se borra el archivo '/home/ragnarok/src/fl2000_drm-rhgndf/fl2000.ko'
make[2]: *** [/usr/src/linux-headers-6.5.0-0.deb12.4-common/Makefile:1977: modules] Error 2
make[1]: *** [/usr/src/linux-headers-6.5.0-0.deb12.4-common/Makefile:246: __sub-make] Error 2
make[1]: se sale del directorio '/usr/src/linux-headers-6.5.0-0.deb12.4-amd64'
make: *** [Makefile:18: modules] Error 2

```

klogg, 2026-10-27:
Closing this historical connector/EDID report as obsolete. Current master has replaced the old bridge implementation with an upstream-based IT66121 driver, changed bridge discovery to operate on the FL2000 adapter's own I2C bus, and reworked DRM setup and hotplug handling. The current supported target is Linux 7.0.

EDID reads, modesetting, visible HDMI output and physical HDMI unplug/reconnect pass on the available hardware. This does not establish that the adapters and Debian configurations in this thread have been retested. "Cannot find any crtc or sizes" alone does not identify the underlying failure; the preceding I2C and connector diagnostics matter.

If the same problem persists with current master on Linux 7.0, please reopen this issue with the exact revision, adapter/output type, USB topology, full kernel log, enumerator service status/log, connector status/EDID and reproduction steps. Please open a separate issue for a different failure. Native VGA is tracked in #79.

Current setup: https://github.com/klogg/fl2000_drm#build
Validation scope: https://github.com/klogg/fl2000_drm/blob/master/docs/DEVELOPMENT.md#validation-scope

closed 2026-10-27 by klogg