klogg/fl2000_drm #59: Linux kernel 6.5.0 support

klogg, 2024-03-21:


klogg, 2024-03-21:
Implemented with compile checks only in 0c87d7c. Need testing on the hardware

ftoledo, 2024-04-13:
can you add support for 6.1 too?
```
ragnarok@ragnarok:~/src/fl2000_drm$ make
make CHECK="/usr/bin/sparse" -C /lib/modules/6.1.0-20-amd64/build M=/home/ragnarok/src/fl2000_drm modules
make[1]: se entra en el directorio '/usr/src/linux-headers-6.1.0-20-amd64'
  CC [M]  /home/ragnarok/src/fl2000_drm/fl2000_drv.o
In file included from /home/ragnarok/src/fl2000_drm/fl2000_drv.c:7:
/home/ragnarok/src/fl2000_drm/fl2000.h:33:10: fatal error: drm/drm_fbdev_generic.h: No existe el fichero o el directorio
   33 | #include <drm/drm_fbdev_generic.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [/usr/src/linux-headers-6.1.0-20-common/scripts/Makefile.build:255: /home/ragnarok/src/fl2000_drm/fl2000_drv.o] Error 1
make[1]: *** [/usr/src/linux-headers-6.1.0-20-common/Makefile:2035: /home/ragnarok/src/fl2000_drm] Error 2
make[1]: se sale del directorio '/usr/src/linux-headers-6.1.0-20-amd64'
make: *** [Makefile:20: modules] Error 2

```

ftoledo, 2024-05-09:
build error on 6.5.0 debian bookworm
```
ragnarok@ragnarok:~/src/fl2000_drm$ make
make CHECK="/usr/bin/sparse" -C /lib/modules/6.5.0-0.deb12.4-amd64/build M=/home/ragnarok/src/fl2000_drm modules
make[1]: se entra en el directorio '/usr/src/linux-headers-6.5.0-0.deb12.4-amd64'
  CC [M]  /home/ragnarok/src/fl2000_drm/fl2000_drv.o
  CC [M]  /home/ragnarok/src/fl2000_drm/fl2000_registers.o
  CC [M]  /home/ragnarok/src/fl2000_drm/fl2000_interrupt.o
  CC [M]  /home/ragnarok/src/fl2000_drm/fl2000_streaming.o
  CC [M]  /home/ragnarok/src/fl2000_drm/fl2000_i2c.o
  CC [M]  /home/ragnarok/src/fl2000_drm/fl2000_drm.o
  LD [M]  /home/ragnarok/src/fl2000_drm/fl2000.o
  CC [M]  /home/ragnarok/src/fl2000_drm/bridge/it66121_drv.o
  LD [M]  /home/ragnarok/src/fl2000_drm/it66121.o
  MODPOST /home/ragnarok/src/fl2000_drm/Module.symvers
ERROR: modpost: "drm_fb_dma_get_gem_obj" [/home/ragnarok/src/fl2000_drm/fl2000.ko] undefined!
ERROR: modpost: "drm_gem_dma_dumb_create" [/home/ragnarok/src/fl2000_drm/fl2000.ko] undefined!
ERROR: modpost: "drm_gem_dma_prime_import_sg_table_vmap" [/home/ragnarok/src/fl2000_drm/fl2000.ko] undefined!
make[3]: *** [/usr/src/linux-headers-6.5.0-0.deb12.4-common/scripts/Makefile.modpost:144: /home/ragnarok/src/fl2000_drm/Module.symvers] Error 1
make[2]: *** [/usr/src/linux-headers-6.5.0-0.deb12.4-common/Makefile:2011: modpost] Error 2
make[1]: *** [/usr/src/linux-headers-6.5.0-0.deb12.4-common/Makefile:246: __sub-make] Error 2
make[1]: se sale del directorio '/usr/src/linux-headers-6.5.0-0.deb12.4-amd64'
make: *** [Makefile:20: modules] Error 2
```

klogg, 2024-05-14:
Please provide (attach) your kernel config. On running system you can do it like [described here](https://superuser.com/questions/287371/obtain-kernel-config-from-currently-running-linux-system)

From the first glance, you have `CONFIG_DRM_GEM_DMA_HELPER ` disabled in your kernel

ftoledo, 2024-05-14:
hi @klogg 

seem to are disabled on debian:

cat /boot/config-6.5.0-0.deb12.4-amd64  |grep CONFIG_DRM_GEM_DMA_HELPER return empty

klogg, 2024-05-14:
Sorry, this is needed for the driver... Can you rebuild the Kerbel for test?

james1mumo, 2024-07-13:
> CONFIG_DRM_GEM_DMA_HELPER

my CONFIG_DRM_GEM_DMA_HELPER seems to be enabled but still am getting the Unknown symbol 

`root@mumo:/home/james/Downloads/fl2000_drm-master# cat /boot/config-$(uname -r) | grep CONFIG_DRM_GEM_DMA_HELPER
CONFIG_DRM_GEM_DMA_HELPER=m
root@mumo:/home/james/Downloads/fl2000_drm-master# insmod fl2000.ko && insmod it66121.ko
insmod: ERROR: could not insert module fl2000.ko: Unknown symbol in module

root@mumo:/home/james/Downloads/fl2000_drm-master# dmesg | tail
[  478.229972] fl2000: Unknown symbol drm_fb_dma_get_gem_obj (err -2)
[  478.230459] fl2000: Unknown symbol drm_gem_dma_dumb_create (err -2)
[  478.230910] fl2000: Unknown symbol drm_gem_dma_prime_import_sg_table_vmap (err -2)

`

Am on ubuntu 22.04, kernel 6.5.0

`###lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.4 LTS
Release:	22.04
Codename:	jammy

###uname -r
6.5.0-41-generic
`

Lassebq, 2025-05-08:
Had the same problem. For some reason it's automatically disabled unless there's an in-tree kernel driver that depends on it

klogg, 2026-10-27:
The project now targets Linux 7.0 rather than maintaining compatibility with Linux 6.5. Current master uses DRM shmem GEM helpers, so the former CONFIG_DRM_GEM_DMA_HELPER dependency discussed here no longer applies. Kconfig selects the required helpers for in-tree integration; external builds prepare the documented missing bridge/HDMI helpers when needed.

Current build instructions and support scope: https://github.com/klogg/fl2000_drm#build

Closing this historical Linux 6.5 support request as outside the current target. This does not claim that Linux 6.5 builds or runs successfully. Please report failures on the documented Linux 7.0 target separately with the exact revision, kernel configuration and complete logs.

closed 2026-10-27 by klogg