klogg/fl2000_drm #71: HDMI Audio

xueyoo, 2024-05-13:
wiki home describe the HDMI Audio not implemented.
is there any plan to implement it?

thanks

klogg, 2024-05-19:
@xueyoo converted into issue

klogg, 2026-10-27:
HDMI audio transport through FL2000 remains unimplemented. The upstream-based IT66121 driver retains its audio support for existing firmware-described boards, but that does not supply an audio transport path for FL2000 adapters.

Remaining scope:
- Establish how audio is transported and wired on the target FL2000 hardware.
- Implement the required Linux audio integration and bridge configuration using existing kernel interfaces where applicable.
- Preserve audio behavior for existing IT6610/21/22 users.
- Validate advertised audio capabilities, playback, video/audio modesets, hotplug, suspend/resume and teardown on suitable hardware.

Current limitations: https://github.com/klogg/fl2000_drm/blob/master/docs/DEVELOPMENT.md#remaining-implementation-work

Keeping this issue open as the FL2000 HDMI audio implementation tracker. No completion date is committed.