klogg/fl2000_drm #72: Display shows as activated, but doesn't work.

TheElevatedOne, 2024-06-06:
So I built it without errors. Signed it, and done all things as they should be.

Now my dmesg shows a bunch of `[  453.163918] usb 2-1: Nonzero urb status, 0`, and I don't know what to do about it.

AlanSce, 2024-06-26:
Hi! 

Same error here, did anyone find a solution?
```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.4 LTS
Release:	22.04
Codename:	jammy

$ uname -r
6.5.0-41-generic
```

Lassebq, 2025-05-09:
It's just a spam of successful USB Request Block transfers. The actual problem seems to be
```
[ 1368.860913] i2c i2c-12: Cannot get monitor status (-5)
```
Tested with a slightly patched driver so that it compiles on kernel 6.14.4

I got it as far as modesetting monitor to the "default" resolution, but monitor edid does not get pulled. Something is wrong in the it66121 driver with the way it accesses i2c or the regmap

Lassebq, 2025-05-09:
I'd be happy to help if the original developer can clear things up, but for now I've hit a roadblock.

klogg, 2026-10-27:
The misleading "Nonzero urb status, 0" logging has been removed. Successful URB completions are handled as success in the current streaming implementation.

The bridge has also been rebased onto the upstream IT66121 driver with transport-specific adaptations. EDID reads, modesetting, visible HDMI output and physical HDMI unplug/reconnect have been tested on the available adapter and display.

Validation and limits: https://github.com/klogg/fl2000_drm/blob/master/docs/HARDWARE.md

Please retest with current master on the documented Linux 7.0 target. If the display still fails, please open a new report with the exact revision, adapter/output type, EDID information and full kernel log. Native VGA remains unimplemented and is tracked in #79.

closed 2026-10-27 by klogg