klogg/fl2000_drm #79: USB3.0 to VGA Adapter

ucycg, 2025-07-15:
Hey I just bought a USB3.0 to VGA Adapter
https://www.amazon.de/GERUI-USB-VGA-Grafikkabel-kompatibel-Schwarz/dp/B0D9GR9LP4?source=ps-sl-shoppingads-lpcontext&ref_=fplfs&smid=A2WSSX5YFMIMA4&th=1

Im using Ubuntu 22.04 and i even downgraded to a Linux 6.5.0 Kernel, but i cant get this driver here to work
is the main reason that this driver is intended for HDMI and NOT Vga? 

when i do lsusb it shows the same FL2000 Chips but probably there is some other component for the VGA output right?
Im just really frustrated after trying for 4hrs so before i send the adapter back i thought i just might ask once :) 

ty in advance

ucycg, 2025-07-15:
Hey man ty so much I was about to return this with Amazon now with the fork I will give it another try :) ty so much couldn't find this via Google. You made my day already for replying ty for your help. I was really frustrated today 

What is the straight forward way to find such forks without Google? :) I'm not really good with git / GitHub yet but I'm so happy to learn

ookiegajwa, 2025-07-15:
On a github repo, there's an 'Insights' tab at the top. From there, you can select 'forks' and see all the forks of a repo. There may be a lot of almost identical repos, but ones with many stars are often useful.

ftoledo, 2026-10-15:
I have working this cheap device using de ACDCS fork 

https://github.com/ADCDS/fl2000drm.git

https://www.mercadolibre.com.ar/up/MLAU190159797?matt_tool=89488245&pdp_filters=item_id:MLA703357266&ua=6AhcxbrYtEwGpKv1TsIbclm3fIW94ovc1i6MkaFDoTQ#origin=share&sid=share&wid=MLA703357266&action=copy

klogg, 2026-10-27:
Native FL2000 VGA output remains unimplemented in this repository. The current display pipeline attaches an external bridge; a VGA-only adapter without that bridge cannot use the HDMI attachment path. Sharing USB ID 1d5c:2000 does not establish the same output hardware.

Keep this issue as the native VGA implementation tracker. Scope:
- Select the native analog output path without requiring an external HDMI bridge.
- Add the appropriate VGA connector, monitor detection and EDID/fallback-mode handling for the actual board wiring.
- Configure analog output and relevant monitor interrupts.
- Validate modes, visible output, hotplug, reset and teardown on VGA hardware.

Thank you for the fork references above. They are useful implementation references, but VGA support from those forks has not been integrated or validated here.

Current limitations: https://github.com/klogg/fl2000_drm/blob/master/docs/DEVELOPMENT.md#remaining-implementation-work